whitecatboard/Lua-RTOS-ESP32 #118: Mqtt can't connect

gjongenelen, 2018-02-21:
Hi,
When connecting to Mqtt it doesn't throw any errors. It just prints
```.MQTT�<{CLIENT_ID}{USERNAME}{PASSWORD}```
With the actual parameters, but omitted here. 

Resulting in the following error after +/- 10 secs:
Guru Meditation Error: Core 0 panic'ed (LoadProhibited)

This used to work before, don't know what happened that it doesn't work now.
Btw, Wifi is connected

jolivepetrus, 2018-02-21:
Hi @gjongenelen,

Tested with last commit, last esp-idf, and last toolchain. Can't reproduce the issue with any secure setup: plain, secured, tls, ....

jolivepetrus, 2018-03-02:
Hi @gjongenelen,

Any progress in this issue? Can be closed?

gjongenelen, 2018-03-05:
It turned out to be a corrupt system, completely wiping the ESP and reflashing it worked. Sorry for the inconvenience!

closed 2018-03-05 by gjongenelen