whitecatboard/Lua-RTOS-ESP32 #119: error http server

mgrusik, 2018-02-27:
https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/8788183c1d45eb38e282b13fd53d7720dc897c3f/components/http/httpsrv.c#L1097

close(client); should be replaced by closesocket(client);

jolivepetrus, 2018-02-27:
Hi @mgrusik,

client socket is binded to the vfs_lwip virtual file system. When you close the socket file descriptor using close the socket is closed, as expected in a POSIX-compliance system.

mgrusik, 2018-02-27:
if you leave it as if that page can only be opened 10 times for the eleventh error

mgrusik, 2018-02-27:
Connections do not close

jolivepetrus, 2018-02-27:
In lwip closesocket is defined as:

static inline int closesocket(int s)
{ return lwip_close_r(s); }

And in vfs_lwip.c close calls to lwip_close_r

    esp_vfs_t vfs = {
        .flags = ESP_VFS_FLAG_DEFAULT | ESP_VFS_FLAG_SHARED_FD_SPACE,
        .write = &lwip_write_r,
        .open = NULL,
        .fstat = NULL,
        .close = &lwip_close_r,
        .read = &lwip_read_r,
        .fcntl = &lwip_fcntl_r_wrapper,
        .ioctl = &lwip_ioctl_r_wrapper,
        .writev = &lwip_writev,
       .select = __real_lwip_select,
    };

... it should be the same to call close or closesocket.

Are you working with last esp-idf, last toolchain, and last Lua RTOS versions?

mgrusik, 2018-02-27:
yes, last versions

mgrusik, 2018-02-28:
sorry, not everything was updated, earned :)

mgrusik, 2018-02-28:
but what's wrong if I have an http server that works if I specify:
CONFIG_LUA_RTOS_HTTP_SERVER_TASK_CPU =0

jolivepetrus, 2018-03-05:
Hi @mgrusik,

It is not apparently any problem running http server on core 0 (just test it), almost if you don't have any other task running on the same core at a highest priority and starving the CPU.

What problem you are experimenting running http server on core 0?

mgrusik, 2018-03-06:
problem with the core 1

jolivepetrus, 2018-03-12:
Hi @mgrusik,

The default sdkconfig file, on all supported platforms run the http server on core 1.
Which is the issue? Any error message?.
Can't reproduce the issue with the information we have.

jolivepetrus, 2018-04-09:
We can't reproduce the issue, and requested info where not provided.
Please, reopen the issue if anyone can give additions information.

closed 2018-04-09 by jolivepetrus