whitecatboard/Lua-RTOS-ESP32 #120: Curl Component conflicts ?

gmenier, 2018-02-28:
Hi, 
Trying to compile for ESPTHING with curl : stops on some redefinitions of struct servent (conflict curl/lib/file.c 52 and lwip/posix/netdb.h:46) and getservbyname - failed to compile file.o for curl.
I just activated curl in menuconfig (all is ok without curl - by the way, luartos = really nice job !). 
What am I missing ?  Perhaps I AM the issue :-)
G.

max1220, 2018-03-03:
I'm having the same issue, using BOARD_TYPE=generic.
After commenting out the relevant piece of code in netdb.h(the redefinition of servent), and disabling some options in curl_config.h, my firmware finally compiled, and I was able to send a simple HTTP get using net.curl.get(). YMMV, this was obviously a hack to "just get it to work".

jolivepetrus, 2018-03-05:
Hi @gmenier, @max1220,

Thanks for report this.

Solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/1c1ed184b5ead3988862e14efe5a0eacb1afbeb3.

closed 2018-03-05 by jolivepetrus