whitecatboard/Lua-RTOS-ESP32 #127: External ADC

wbvreeuwijk, 2018-03-21:
Hi,

I've worked with the ESP8266 for quite some time using NodeMCU Lua and am now starting to move to ESP32. I find the Lua-RTOS an excellent implementation and am starting to experiment with it. I'm trying to catch up on all the documentation and have started working with the ADC module. I have an ADS1115 module and wanted to work with it. Following the documentation I started with the following code;

`tmp = adc.attach(adc.ADS1115,adc.ADC_CH0)`

This resulted in the following error: 

`stdin:1: bad argument #1 to 'attach' (number expected, got nil)`
`stack traceback:`
`        [C]: in field 'attach'`
`        stdin:1: in main chunk`
`        [C]: in ?`

I ran the `_info.lua` code and found
`"externalADC":{  
      "MCP3008":false,
      "MCP3208":false,
      "ADS1115":false,
      "ADS1015":false
   },`

This would suggest that the module is not enabled. How do I enable this module? Is it through code or through flashing different firmware?

Thanks in advance,
Bas

jolivepetrus, 2018-03-21:
Hi @wbvreeuwijk,

You must build Lua RTOS with ADS1115 support, your build doesn't include ADS1115 support.
You can build Lua RTOS with the modules you need in your setup to save flash space.

To configure Lua RTOS you must run "make menuconfig" first. To enable ADS1115 support you must go to:

Lua RTOS -> Hardware -> External ADC -> External ADC type and enable ADS1115:

[image]

Regards,

Jaume

wbvreeuwijk, 2018-03-22:
Thanks for the quick reply Jaume.

Yes I was afraid of that :-)

I setup the toolchain and build enviroment but now get an error during compilation: 

`/firmware/Lua-RTOS-ESP32/components/lua/modules/bluetooth/bluetooth_eddystone.inc:201: undefined reference to `hex_string_to_val'`

I assume that this is due to a difference in Espressif SDK version. I used this https://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html to setup the environment.

Other then disableing bluetooth is there something else I can do?

jolivepetrus, 2018-03-22:
Hi @wbvreeuwijk,

Which board type in your build?

Please, enter your board type:

 1: Whitecat N1 ESP32
 2: Whitecat N1 ESP32 with OTA
 3: Whitecat N1 ESP32 DEVKIT
 4: Whitecat N1 ESP32 DEVKIT with OTA
 5: Whitecat ESP32 LORA GW
 6: Whitecat ESP32 LORA GW With OTA
 7: Espressif Systems ESP32-CoreBoard
 8: Espressif Systems ESP32-CoreBoard with OTA
 9: Espressif Systems ESP-WROVER-KIT
10: Espressif Systems ESP-WROVER-KIT with OTA
11: SparkFun ESP32 Thing
12: SparkFun ESP32 Thing with OTA
13: Olimex ESP32-Gateway
14: Olimex ESP32-Gateway with OTA
15: Olimex ESP32-EVB
16: Olimex ESP32-EVB with OTA
17: Generic ESP32 board
18: Generic ESP32 board with OTA

wbvreeuwijk, 2018-03-22:
I'm building for Espressif Systems ESP32-CoreBoard.
I managed to build the image by disabling bluetooth.

jolivepetrus, 2018-03-22:
@wbvreeuwijk,

Please, try with last commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/a4b96e7361f54d2146c10ea791c6db0959832fc1 to enable bluetooh support also.

wbvreeuwijk, 2018-03-24:
I was able to build the ROM but when I had flashed it it didn't work

wbvreeuwijk, 2018-03-26:
I was able to build and run the image by disabling Bluetooth. When Bluetooth was enabled the module would get in a bootloop.

closed 2018-03-26 by wbvreeuwijk