whitecatboard/Lua-RTOS-ESP32 #128: Eddystone example not working

webfrank, 2018-03-23:
Hi, I’ve tried the example and the esp32 is detected but I cannot provision the beacon with google beacon tool.  Also the physical web does not receive the url notification.

What could be the issue?

jolivepetrus, 2018-04-09:
Hi @webfrank,

Any error message?

webfrank, 2018-04-10:
Hi, no error messages, but after some tries I figured out how to push an url which should be in https and not an url shortener. The real problem is that the google beacon tool does not permit eddystone configuration, probably it is looking for a configuration service which is not implemented.

the0ne, 2019-04-09:
@webfrank as the issue is fixed for you, would you be able to provide a short howto? We could integrate that into the wiki and it would help others a lot.

the0ne, 2019-07-26:
@webfrank please provide a short howto - thanks in advance!
Closing this issue for now.

closed 2019-07-26 by the0ne

the0ne, 2019-07-26:
Using either of the samples

https://github.com/whitecatboard/Lua-RTOS-ESP32/tree/master/components/fs_images/default/examples/lua/eddystone.lua

```
b1 = bt.service.eddystone.uid("808182838485", 0xed, "3F3B51B60B88EF9949E5", "CF484185AE0B")
b2 = bt.service.eddystone.url("808182838485", 0xed, "https://whitecatboard.org")

b1:start()
b2:start()
```

https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/components/fs_images/default/examples/lua/ble-eddystone.lua

```
b1 = bt.service.eddystone.uid("808182838485", -42, "3F3B51B60B88EF9949E5", "CF484185AE0B")
b2 = bt.service.eddystone.url("808182838485", -42, "https://whitecatboard.org")

b1:start()
b2:start()
```

I could easily create a working eddystone beacon.