whitecatboard/Lua-RTOS-ESP32 #129: Brownout detection was triggered when starting WiFi

wbvreeuwijk, 2018-03-25:
When starting `net.wf.scan()` I get the error

`Brownout detector was triggered

ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:10212
entry 0x40078a00`

lu1210, 2018-03-25:
I had the same problems.  A  1000 yF elko placed near the ESP32 Vin and GND pins solved the poblem.

jolivepetrus, 2018-03-26:
Hi @wbvreeuwijk,

With which board?

wbvreeuwijk, 2018-03-26:
It's a  TTGO wemos MINI D1 ESP32 board. I use the Core 32 image.
One board had no problem, but this one does. I managed to build an image with without brownout detection and it works, I will check the suggestion by @lu1210 .

Strangely enough, I have not been successful in building images using this repository but was able to build a working image using the fork by @the0ne using the same workflow. I will check where the differences are. To be clear building the image works, but the module enters a boot-loop.

the0ne, 2018-03-27:
I believe using my fork doesn't have a real influence. Might be an incident only.
I propose you use a *really good* USB cable. Try several.
I also have cables that show issues delivering enough power when switching on wifi.
The alternative would be to add a capacitor on the board - but I've never done that myself.

jolivepetrus, 2018-04-05:
Hi guys,

As commented above, this is a hardware related issue, so we close it.

closed 2018-04-05 by jolivepetrus