whitecatboard/Lua-RTOS-ESP32 #137: I2c script with READ goes wrong

lu1210, 2018-04-06:
Following i2c script goes wrong. The used chip is a BMP280 .
The same problem exists on a second chip MCP4725 also.

/ > x:start()
/ > x:address(0x76,false)
/ > print('WRITE 0xD0',x:write(0xd0))   -- get chip id
WRITE 0xD0                               
/ > print('Stop',x:stop())
Stop                                     -- ok
/ > x:start()
/ > x:address(0x76,true)
/ > print('READ',string.format("H'%02X",x:read()))  -- read chip id
READ	H'58                                          -- ok, chipid = BMp280
/ > x:stop()                                         -- stop goes wrong !!!!!
stdin:1: 50331655:timeout
stack traceback:
	[C]: in method 'stop'
	stdin:1: in main chunk
	[C]: in ?
/ >

jolivepetrus, 2018-04-07:
@lu1210,

Please, try:

x:start()
x:address(0x76,false)
x:write(0xd0)
x:start()
x:address(0x76,true)
x:read()
x:stop()

To read a register from BMP280 the sequence is "start -> address for write -> write -> start -> address for read -> read -> stop", not start -> address for write -> write -> stop-> start -> address for read -> read -> stop ".

Reference:

https://cdn-shop.adafruit.com/datasheets/BST-BMP280-DS001-11.pdf

pages 29 & 30.

lu1210, 2018-04-07:
/ > if not bmp then bmp = i2c.attach(i2c.I2C0,i2c.MASTER) end
/ > bmp:start()
/ > bmp:address(0x76,false)
/ > bmp:write(0xd0)
/ > bmp:start()
/ > bmp:address(0x76,true)
/ > bmp:read()
88
/ > bmp:stop()
stdin:1: 50331655:timeout
stack traceback:
	[C]: in method 'stop'
	stdin:1: in main chunk
	[C]: in ?
/ > 
/ > if not bmp then bmp = i2c.attach(i2c.I2C0,i2c.MASTER) end
/ > bmp:start()
/ > bmp:address(0x76,false)
/ > bmp:write(0xd0)
/ > bmp:start()
/ > bmp:address(0x76,true)
/ > bmp:read()
88
/ > bmp:stop()
stdin:1: 50331655:timeout
stack traceback:
	[C]: in method 'stop'
	stdin:1: in main chunk
	[C]: in ?
/ > 
/ > if not bmp then bmp = i2c.attach(i2c.I2C0,i2c.MASTER) end
/ > bmp:start()
/ > bmp:address(0x76,false)
/ > bmp:write(0xd0)
/ > bmp:start()
/ > bmp:address(0x76,true)
/ > bmp:read()
88
/ > bmp:stop()
stdin:1: 50331655:timeout
stack traceback:
	[C]: in method 'stop'
	stdin:1: in main chunk
	[C]: in ?
/ >

jolivepetrus, 2018-04-09:
@lu1210,

Solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/b669d373934c9eeff8003fabeec29c11e9863a90.

It will be available in the precompiled firmware images tomorrow.
Available for builds from the scratch.

lu1210, 2018-04-09:
Problem is solved with new build.

Thank You very much for the quick response

closed 2018-04-09 by lu1210