whitecatboard/Lua-RTOS-ESP32 #138: MQTT publish with QoS > 0 seems it doesn't working

piebat, 2018-04-07:
MQTT publish with QoS > 0 seems doesn't work. 
I tried to publish the message with QoS = 1 and 2, but this message raise:

>  201326596:can't publish to topic (enable persistence for a qos > 0)

I couldn't find documentation about how to change the persistence and seems that default is:
MQTTCLIENT_PERSISTENCE_NONE instead than MQTTCLIENT_PERSISTENCE_DEFAULT.

The environment is:
Lua RTOS beta 0.1. build 1523007799 commit 5d9b98a49aeb0a799c3949ba9bf904f591963756
Running from factory partition
board type ESP32COREBOARD
cpu ESP32 rev 1 at 240 Mhz
spiffs0 start address at 0x310000, size 512 Kb, partition spiffs
spiffs0 mounted

With QOS0 work fine.

jolivepetrus, 2018-04-09:
@bytebang,

Thanks for the documentation issue. It has been updated.

piebat, 2018-04-10:
@bytebang 
Thanks Ghunter, work fine.

closed 2018-04-10 by piebat