whitecatboard/Lua-RTOS-ESP32 #139: UPTIME is missing.

oskarirauta, 2018-04-08:
Lua-RTOS-ESP32 does not seem to have any "uptime" capability.
This should be made available (seconds/time from boot) - I made some testing with new board and was wondering if it was running smoothly for the whole night, and decided to check this at morning, but unfortunately there was no uptime command in shell (nor did I find anything similar for Lua)

bytebang, 2018-04-08:
I think the following function should do the trick: https://www.freertos.org/a00021.html#xTaskGetTickCount

oskarirauta, 2018-04-08:
Found it aswell- also something useful maybe in curl’s timeval.c, but I was not able to implement this...

Lähetetty iPhonesta

gjongenelen, 2018-04-08:
I think you want to use **os.clock()**?

This return the amount of seconds counted since boot. Couldn't find any documentation about this command though.

oskarirauta, 2018-04-09:
Perfect. Although, I still think it should be mapped to uptime command in shell.

Lähetetty iPhonesta

jolivepetrus, 2018-04-09:
Hi guys,

This feature will be added as soon as possible.

oskarirauta, 2018-04-09:
Thanks. Not a big deal, but it is improvement definetly.

Lähetetty iPhonesta

jolivepetrus, 2018-04-09:
Hi guys,

Added in commit https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Lua-RTOS-&-board-status.

jolivepetrus, 2018-04-09:
Hi guys,

Added in commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/050455de70a6b685642a75918db658b18ef10e3c.

Wiki: https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Lua-RTOS-&-board-status#osuptimetable

closed 2018-04-09 by jolivepetrus

oskarirauta, 2018-04-21:
Just tested this. Works perfectly. Thanks.