whitecatboard/Lua-RTOS-ESP32 #140: Needed time for an I2C:STOP isn't  plausible

lu1210, 2018-04-12:
Timing of I2C:STOP of  an I2C:READ isn't plausible. It needs about 1 second!??!
This measurements were done in a test configuration with BMP280 and MCP-4725(DAC)

The timing after an I2C:WRITE is ok.

write 1 byte into registor
START	nil	0.000 sec
ADDRESS	nil	0.001 sec
WRITE	nil	0.000 sec
WRITE	nil	0.000 sec
STOP	nil	0.001 sec     <---- OK
end write 1 byte into register
read register 1 byte
START	nil	0.000 sec
ADDRESS	nil	0.000 sec
WRITE	nil	0.000 sec
STOP	nil	0.000 sec     <---- OK
START	nil	0.000 sec
ADDRESS	nil	0.000 sec
READ:	H' 7E	0.000 sec
STOP	nil	1.000 sec     <---- not OK
End read 1 byte

I can not use a multiple- or burst READ also, because I don't know the correct syntax
(is burst READ implemented already).
This would reduce interface traffic and time.

jolivepetrus, 2018-04-12:
Hi @lu1210,

Many thanks for your tests on I2C Lua module!!

Please, try with https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/7b98bff966fc870add6122d1da568e943b20ec40. A patch is needed in I2C esp-idf driver to deal with the Lua RTOS I2C Lua module.

Sorry for the inconvenience.

lu1210, 2018-04-16:
add 1) single read and timeing of method stop is with patch 7b98bff ok, now!

End read 1 byte
read register 1 byte
START	nil	0.001 sec
ADDRESS	nil	0.000 sec
WRITE	nil	0.000 sec
STOP	nil	0.000 sec
START	nil	0.000 sec
ADDRESS	nil	0.001 sec
READ:	H' 71	0.001 sec
STOP	nil	0.001 sec   < -- ok

add 2) Multiple or burtst read isn'nt possible
- 1st READ is ok
-- 2nd READ returns always h'FF

/ >       if not hi2c then hi2c = i2c.attach(i2c.I2C0,i2c.MASTER,400000) end
/ >       hi2c:start()
/ >       hi2c:address(0x62,true)
/ >       print('1',require('bin')(hi2c:read(3)))
1	H'C0 B'1100-0000
/ >       print('2',require('bin')(hi2c:read()))
2	H'FF B'1111-1111
/ >       print('3',require('bin')(hi2c:read()))
3	H'FF B'1111-1111
/ >       print('4',require('bin')(hi2c:read()))
4	H'FF B'1111-1111
/ >       hi2c:stop()

jolivepetrus, 2018-04-16:
@lu1210,

This is solved in this commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/97bf247d6faf4c29caaa7bb64efbd2515e2bccfe.

Please, try again.

I hope the solution is near.

;)

lu1210, 2018-04-16:
Patch 97bf247 solves the problem(not only this one :-). 
It was tested successfully with a MCP4725.
 > writeD
          AC(1000,true)
/ > readDAC()
1	H'C0 B'1100-0000
2	H'FF B'1111-1111
3	H'F0 B'1111-0000
0
/ >       if not hi2c then hi2c = i2c.attach(i2c.I2C0,i2c.MASTER,400000) end
/ >       hi2c:start()
/ >       hi2c:address(0x62,true)
/ >       print('1',require('bin')(hi2c:read(3)))
1	H'C0 B'1100-0000
/ >       print('2',require('bin')(hi2c:read()))
2	H'FF B'1111-1111
/ >       print('3',require('bin')(hi2c:read()))
3	H'F0 B'1111-0000
/ >       hi2c:stop()

closed 2018-04-16 by lu1210