whitecatboard/Lua-RTOS-ESP32 #142: 234881027:cannot start thread (No such file or directory)

lu1210, 2018-04-17:
Hi,
since last update i got this error message, sometime(?!)

12:46:38 printKD not started 234881027:cannot start thread (No su
         ch file or directory)
f	    1537		printKD.lua
f	     908		genKD.lua
f	     485		genKD.lc
f	     941		printKD.lc  <-- file does exists

if thread.status(tprintKD) ~= 'running' then
   succ, tprintKD = pcall(thread.start,
   function()
      local suc, msg = pcall(dofile,'printKD.lc')
      if not suc then tprint(4,'printKD:',msg) end
   end)
   if succ then
      tThread.printKD = tprintKD
   else
      tprint(4,'printKD not started',tprintKD)
      os.ls('*KD.*')
      tprintKD = 0
   end
end

jolivepetrus, 2018-04-18:
Mmmm .....

Which board? Using PSRAM?

lu1210, 2018-04-18:
I am using board ESP32 DEVKIT V1 from DOIT.
Due the fact I don't know what PSRAM is, I think I have no PSRAM :- )

jolivepetrus, 2018-04-18:
Hi @lu1210,

Can't reproduce. Please, provide working example script that causes the issue in your board for testing.

lu1210, 2018-04-19:
HI, 
I'm sorry, but I cann't reproduce the error with a simple script. Can I send the whole scrips in a zip file to you?

jolivepetrus, 2018-04-19:
Yes, of course.

Send to jolive@whitecatboard.org

lu1210, 2018-04-20:
it seems, it works perfect. Congratulation!!!

lu1210, 2018-04-20:
I am sorry. the error isn't solved :-(

17:56:56 printKD not started 234881027:cannot start thread (No su
         ch file or directory)
f	     941		printKD.lc
f	    1292		printKD.lua
Free mem: 56072

lu1210, 2018-04-20:
By the way I don't understand the output of this script!
Can you try this one.
thread.start(
function()
--  local succ,msg,tsucc,tmsg
    while true do
       tid = thread.start(
       function()
           succ,msg = pcall(dofile,"test.lua")
           if not succ then print(msg)
              print('II',thread.status(tid))
            end  
         end)
      tmr.delay(10)
      thread.list()
    end  
 end)
 
there are always same THID's ?????????????
No entiendo nada

lu1210, 2018-04-20:
next I don't understand!

thread.start(function() print('HUBER') end,16384,20,1)
stdin:1: 234881027:cannot start thread (No such file or directory)
stack traceback:
	[C]: in field 'start'
	stdin:1: in main chunk
	[C]: in ?

jolivepetrus, 2018-04-22:
Hi @lu1210,

About your question about that thid are equal ...

Extracted from posix 1.1 pthread spec:

_**Thread IDs are guaranteed to be unique only within a process.  A thread ID may be reused after a terminated thread has been joined, or a detached thread has terminated.**_

The thing is that when you create the new thread every 10 seconds probably the previous thread is yet terminated and the thread id is reused. Internally, in the Lua RTOS pthread implementation the thread id is a pointer to a pthread structure created in the heap and released when the thread terminates, so the is a probability that when you create the new thread the same block is the heap is reused.

Other previous issues has been resolved in:

https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/7ea5a4373b95e6a9fc621316b5216758cdec0d0f

https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/99c4be328bb788bfa0def8616aa7656cc4f33b4a

lu1210, 2018-04-22:
You are right, but in my understanding, there could'nt be the sane THID at the same time(see example) or thread.list goes wrong.

-----------------------------------------------------------------------------------------------
           |        |                  |        |      |      |            STACK               
THID       | TYPE   | NAME             | STATUS | CORE | PRIO |   SIZE     FREE     USED       
-----------------------------------------------------------------------------------------------
1073569396   lua      Audit              run         0     20    10216     5520     4696 ( 45%)   
1073478684   lua      lua_main           run         0     20    20480    12004     8476 ( 41%)   
1073458328   lua      Hackler            run         1     20    16384     8036     8348 ( 50%)   
1073501232   lua      Receive            run         1     21     8192     6544     1648 ( 20%)      < ----------
1073501232   lua      Recovery           run         1     17    16384    11952     4432 ( 27%)   < ----------
1073601992   lua      lua_thread         run         1     20    16384    13668     2716 ( 16%)   
1073517396   lua      lua_thread         run         1     20    16384    13672     2712 ( 16%)   
1073515572   lua      lua_thread         run         1     20    16384    14600     1784 ( 10%)   
1073462516   lua      lua_thread         run         1     20    16384    12460     3924 ( 23%)

jolivepetrus, 2018-04-23:
@lu1210,

Please, try again with the last commit. The only apparent reason for the duplicate 1073501232 thid is that one of the threads already is deleted, but it's TCB is not free yet by FreeRTOS.

In recent commits deleted TCB are not considered in list.

lu1210, 2018-04-23:
hi jolivepetrus,
with new firmware update all works perfect:-)

closed 2018-04-28 by lu1210