whitecatboard/Lua-RTOS-ESP32 #149: Internal LED control

oskarirauta, 2018-04-21:
I think we should be ablel to control internal LED (if set) - or atleast, stop blinking.
Currently this can be archieved by setting it's GPIO state to input, but I think there could be a better way as well.

How about a minimal module for this?
led state module for selecting mode from manual or automatic (blinking).

```
led.mode(LED.manual) -- stop blinking (and turn it off)
led.state(LED.on) -- turn it on if manual mode is enabled
led.state(LED.off) -- turn it off
led.mode(LED.automatic) -- resume normal operation mode (start blinking)
```

jolivepetrus, 2018-04-22:
Hi @oskarirauta,

Something about your request is done in the mesh feature that we will add to Lua RTOS soon. There will be a new Lua function in the os module to control the activity led.

oskarirauta, 2018-04-22:
Sounds good :)

the0ne, 2019-07-26:
@oskarirauta as you can see the LED you mention is not built into the esp32 chip itself but into the board that the esp32 is mounted on:

[image]
So - depending on which PIO the board designer connected the LED to - controlling the LED is as easy as:

```
-- this depends on the board
led_pin = pio.GPIO2
-- enable controlling the LED
pio.pin.setdir(pio.OUTPUT, led_pin)
-- enable the on-board LED
pio.pin.sethigh(led_pin)
-- disable the on-board LED
pio.pin.setlow(led_pin)
```

@jolivepetrus do you want to keep this open?