whitecatboard/Lua-RTOS-ESP32 #151: Can't access SPI peripheral.

yawor, 2018-04-26:
Hi,

I have a ai-thinker.com esp32s devkit board (here's electrical schcematic http://pliki.propix.com.pl/images/nodemcu_esp-32s/nodemcu_32s_sch.png) with 4 MB of flash memory.

I've been able to successfully install Lua RTOS on it using either "Whitecat N1 ESP32 DEVKIT" or "Generic ESP32 board". The Lua RTOS boots, I'm able to connect to serial console and I've been able to configure wifi connection and ssh server.

But I'm trying to use SPI and I'm getting errors.

On "Whitecat N1 ESP32 DEVKIT", when trying to attach to SPI2:
```
/ > s = spi.attach(spi.SPI2, spi.MASTER, pio.GPIO5, 50000000, 8, 0)
assertion "ret==ESP_OK" failed: file "/root/esp/Lua-RTOS-ESP32/components/sys/drivers/spi.c", line 653, function: spi_ll_setup
abort() was called at PC 0x400d47a7 on core 0

Backtrace: 0x40095a77:0x3ffd86d0 0x40095bcf:0x3ffd86f0 0x400d47a7:0x3ffd8710 0x400e05d7:0x3ffd8740 0x400e13df:0x3ffd87b0 0x400ed8bc:0x3ffd87f0 0x400fe7e5:0x3ffd8830 0x40101df1:0x3ffd8860 0x400fea41:0x3ffd88b0 0x400fea61:0x3ffd88d0 0x40103a41:0x3ffd88f0 0x400fe415:0x3ffd8910 0x400febea:0x3ffd8990 0x401048ad:0x3ffd89c0 0x400fdd30:0x3ffd89f0 0x400fe00e:0x3ffd8a10 0x400fe24a:0x3ffd8a40 0x400fe7e5:0x3ffd8a70 0x400fea3a:0x3ffd8aa0 0x400fea61:0x3ffd8ac0 0x40103a41:0x3ffd8ae0 0x400fe415:0x3ffd8b00 0x400febea:0x3ffd8b80 0x401048ad:0x3ffd8bb0 0x400fdfc4:0x3ffd8be0 0x400d5797:0x3ffd8c10 0x400d542f:0x3ffd8c30
```

On "Whitecat N1 ESP32 DEVKIT" and "Generic ESP32 board", with SPI3:
```
/ > s = spi.attach(spi.SPI3, spi.MASTER, pio.GPIO5, 50000000, 8, 0)
spi3 at pins miso=GPIO19/mosi=GPIO23/clk=GPIO18
/ > s:select()
stdin:1: 83886087:invalid device
stack traceback:
	[C]: in method 'select'
	stdin:1: in main chunk
	[C]: in ?
/ > s:write(1)
stdin:1: 83886087:invalid device
stack traceback:
	[C]: in method 'write'
	stdin:1: in main chunk
	[C]: in ?
/ > s:deselect()
stdin:1: 83886087:invalid device
stack traceback:
	[C]: in method 'deselect'
	stdin:1: in main chunk
	[C]: in ?
```

On "Generic ESP32 board", I'm able to attach to SPI2, but I also get "invalid device" error when trying to use it (like with SPI3 above).

This happens on both the official releases and on self-built ones (using defconfig and then changing only flash programmer parameters).

xopxe, 2018-04-26:
We have the same problem with a SparkfunESP32  Thing, using a image built from sources.

`/ > device = spi.attach(spi.SPI2, spi.MASTER, pio.GPIO15, 1000000, 8, 0)
spi2 at pins miso=GPIO12/mosi=GPIO13/clk=GPIO14
/ > device:select()
stdin:1: 83886087:invalid device
stack traceback:
	[C]: in method 'select'
	stdin:1: in main chunk
	[C]: in ?
`

yawor, 2018-04-26:
I've started tracking the issue. The error is generated in `components/sys/drivers/spi.c` by checks in several functions like this:
```c
        if ((device < 0) || (device > SPI_BUS_DEVICES)) {
                return driver_error(SPI_DRIVER, SPI_ERR_INVALID_DEVICE, NULL);
        }
```
where `SPI_BUS_DEVICES` is defined as value `3`.

`device` is a number returned by function `spi_get_free_device`, called by `spi_ll_setup`:
```c
static int spi_get_free_device(int unit) {
        int i;

        for(i=0;i < SPI_BUS_DEVICES;i++) {
                if (!spi_bus[spi_idx(unit)].device[i].setup) return i;
        }

        return -1;
}
```
I think the value returned here must be greater or equal to `3` for some reason. It shouldn't be `-1` because it would error out sooner than call to `spi:select()` or other functions.

yawor, 2018-04-26:
OK, it's not the above check of `device` value against `SPI_BUS_DEVICES`. The value is ok. It's the next check:
```c
        if (!spi_bus[spi_idx(unit)].device[device].setup) {
                return driver_error(SPI_DRIVER, SPI_ERR_DEVICE_NOT_SETUP, NULL);
        }
```
The error message is the same for `SPI_ERR_DEVICE_NOT_SETUP` and `SPI_ERR_INVALID_DEVICE`. It seems that the `setup` field in `spi_device_t` struct is not being properly set to `1`.

yawor, 2018-04-26:
OK, the device's setup field is not set anywhere in the code. I'll post a PR in a moment which fixes this. I've tested SPI with a logic analyser and it works.
BTW when I do this:
```lua
s:select()
s:write(1, 2, 3, 4, 5)
s:deselect()
```
I see in the logic analyser that each single byte is sent quickly (according to the frequency set in the `spi.attach()` (it takes about 1.5 us at 5 MHz), but the delay between bytes is really big, even up to 37 us. It means that sending 5 bytes like above takes 5 bytes * 1.5 us + 4 delays * 36 us (average) = 151 us instead just about 8 to 10 us if there were no such delays. Also time from select to first byte and time from last byte to deselect are really large: 170 us and 81 us respectively.

yawor, 2018-04-26:
Captured SPI signal when sending string "abcde" to show timing issue mentioned in my last comment:
[image]

jolivepetrus, 2018-04-28:
Please, try with last commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/3ccc656f52fead714be34414359b9c797d769498

that solves the following:

/ > s = spi.attach(spi.SPI2, spi.MASTER, pio.GPIO5, 5000000, 8, 0)
assertion "ret==ESP_OK" failed: file "/root/esp/Lua-RTOS-ESP32/components/sys/drivers/spi.c", line 653, function: spi_ll_setup
abort() was called at PC 0x400d47a7 on core 0

jolivepetrus, 2018-04-29:
Hi @yawor,

About the timing issues, please check this commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/936b6f0bf8e612b04d80d1ed4405e1a06d01ae65. Now the read/write data is stored into a buffer and spi bulk functions are used. This should speed up the transfers.

yawor, 2018-04-30:
@jolivepetrus thanks. I can confirm that the transaction is now a little faster, without unnecessary pauses between bytes.

Unfortunately this doesn't help as the times from activating NSS line to first clock signal is still really big. I'm trying to interface with 868 MHz RFM12B module, which doesn't have any internal buffering on receiver side (it only stores a single received byte), so the timing is really strict. If the received byte is not read quickly enough, it gets overwritten (this situation can be detected by checking a status register, but this doesn't solve the issue in any way).

The module has an output nIRQ pin, which pulls down the line when something important is happening (like next byte received). The program needs to react to the interrupt by first requesting a status register and if there's a byte received, issuing a read command.

I've set up the interrupt by using (undocumented?) `pio.pin.interrupt` function to trigger only on falling edge. Unfortunately I'm getting a lot of missed bytes.

The incoming data has a baud rate of 9600 bps. This means I only have about 0.8 ms to react to an interrupt, check the status and read the byte before the next one rolls in.

I did some additional measurements:
1. time from the interrupt signal to activating NSS (status command): 0.83 ms
2. time from activating NSS to first SPI clock edge: 0.1 ms
3. time from last clock edge to deactivating NSS: 0.13 ms (about 1 ms passed from the interrupt and we've only read status register so far)
4. then there's a small delay (~ 50 us)
5. similar timings to 2 and 3 when issuing read command

I wonder why the delay between SPI transactions (50 us) is much smaller than the delays inside the transaction. Also the reaction time to interrupt alone can be responsible for missing one or two bytes at the beginning.

Is there anything that can be done to optimise interrupt reaction times and SPI operations or is the Lua interpreter just too slow for this? Do I need to try to write the low level stuff as a module in C instead? I'd like to have that Real Time from the Lua-RTOS name :). Both ESP32 and Lua are quite new to me, but I'm a quick learner, so maybe I could help. Just point me in the right direction.

Here's a code I'm writing, for reference:
```lua
-- RFM12B

local FREQUENCY = 1664 -- math.floor((868320000 - 860000000) / 5000) -- 868.320 MHz
local FSK_SHIFT = 48 -- (math.floor(60000 / 15000) - 1) << 4 -- 60000
local DATARATE = 35 -- math.floor((10000000 / 29 / 9600) - 0.5)

local CMD_CFG = 0x8000
local CFG_EL = 0x80
local CFG_EF = 0x40
--local CFG_BAND_315 = 0x00
--local CFG_BAND_433 = 0x10
local CFG_BAND_868 = 0x20
--local CFG_BAND_915 = 0x30
--local CFG_XTAL_8_5PF  = 0x00
--local CFG_XTAL_9_0PF  = 0x01
--local CFG_XTAL_9_5PF  = 0x02
--local CFG_XTAL_10_0PF = 0x03
--local CFG_XTAL_10_5PF = 0x04
--local CFG_XTAL_11_0PF = 0x05
--local CFG_XTAL_11_5PF = 0x06
--local CFG_XTAL_12_0PF = 0x07
local CFG_XTAL_12_5PF = 0x08
--local CFG_XTAL_13_0PF = 0x09
--local CFG_XTAL_13_5PF = 0x0A
--local CFG_XTAL_14_0PF = 0x0B
--local CFG_XTAL_14_5PF = 0x0C
--local CFG_XTAL_15_0PF = 0x0D
--local CFG_XTAL_15_5PF = 0x0E
--local CFG_XTAL_16_0PF = 0x0F

local CMD_PWRMGT = 0x8200
local PWRMGT_ER = 0x80
--local PWRMGT_EBB = 0x40
local PWRMGT_ET = 0x20
--local PWRMGT_ES = 0x10
--local PWRMGT_EX = 0x08
--local PWRMGT_EB = 0x04
--local PWRMGT_EW = 0x02
local PWRMGT_DC = 0x01

local CMD_FREQUENCY = 0xA000

local CMD_DATARATE = 0xC600
--local DATARATE_CS = 0x80

local CMD_RXCTRL = 0x9000
local RXCTRL_P16_VDI = 0x400
[... 155 lines cut ...]

function rfm12_callback()
    if disableCB then return end
    disableCB = true
    while true do
        status = rfm12_readwrite(CMD_STATUS)
        recheck = false
        if status & STATUS_FFIT ~= 0 then
            recheck = true
            recvd = recvd + 1
            buffer[recvd] = rfm12_readwrite(CMD_READ) & 0xff
            if recvd == 11 then
                rfm12_readwrite(CMD_CLEAR_FIFO)
                rfm12_readwrite(CMD_ACCEPT_DATA)
                table.insert(packets, buffer)
                buffer = {}
                recvd = 0
                packet_rcvd:broadcast()
            end
        end
        if not recheck then break end
    end
    disableCB = false
end
```

jolivepetrus, 2018-04-30:
@yawor,

It should feasible to do that from Lua. Please, try with commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/8e2a39b1a5cba6450c341e21f7afd15ba7fd8e6b, that solves some timing issues.

yawor, 2018-04-30:
@jolivepetrus it much much better now. The reaction to the interrupt now went from around 830 us down to just about 73 us. That's 10 times improvement :).
I still see some receive errors (malformed packets), but I need to investigate if it's an algorithm problem or the times are still too loose - the time the NSS is being active is the only thing left untouched.

BTW what happens when another interrupt is generated when the callback started by the previous one is still executing? Is it queued and calls the callback when previous finishes? Or starts another thread with another callback? Or maybe it's automatically ignored?

When I started sending packets more frequently (about 10-12 in a second, each packet is 11 bytes long) I started getting:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x400f3eb5  PS      : 0x00060d30  A0      : 0x800f4398  A1      : 0x3ffed140  
A2      : 0x3ffd9600  A3      : 0x3ffd969c  A4      : 0x3ffe4e28  A5      : 0x3ffd9600  
A6      : 0x3ffdbd64  A7      : 0x00000004  A8      : 0x00000002  A9      : 0x3ffe4e44  
A10     : 0x3ffd9c08  A11     : 0x000000ff  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0x00000040  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400012e5  LEND    : 0x40001309  LCOUNT  : 0x800fdacb  

Backtrace: 0x400f3eb5:0x3ffed140 0x400f4395:0x3ffed170 0x400ff68a:0x3ffed190 0x40100275:0x3ffed1c0 0x400f4d15:0x3ffed210 0x400f4d35:0x3ffed230 0x4010268c:0x3ffed250 0x400eb405:0x3ffed270
```

jolivepetrus, 2018-04-30:
@yawor,

Do you use the gdisplay code in your code?

yawor, 2018-05-01:
@jolivepetrus no. There's nothing more than the code I've pasted in my previous comment. I only have the RFM12B connected (SPI full duplex + one GPIO for IRQ and one GPIO in output mode for RFM12B reset).
I don't even have any network services running right now (wifi is not enabled). In autorun.lua I only load the rfm12b.lua and call the rfm12_init function.

jolivepetrus, 2018-05-01:
@yawor,

Ok, please try with https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/f9328aa43af1578088c63dca797a49567f9c0690.

The Lua interrupt management is based in the deferred interrupt concept, where the callback is executed outside the ISR (to resolve, for example, stack size issues), but a real ISR is simulated. This means that once the interrupt is received and queued, the GPIO ISR is detached and attached again when the Lua callback finish; so there is a possibility to loss interrupts. This was is due to an arbitrary design decision, that we can remove.

yawor, 2018-05-01:
@jolivepetrus on the f9328aa43af1578088c63dca797a49567f9c0690 I now only see a sporadic receive errors (I can work with that, the protocol I'm working repeats each packet 2-3 times anyway). Also I've spammed the receiver with a ton of packets in a short time and I didn't see guru meditation anymore.
I don't see any noticeable difference with d70cdfe544239955a702c6aaaf9c2a375c2caa55, but looking at the commit, it seems to be important too.

Regarding the ISR being disabled during callback, I think it's reasonable. The original code I'm porting the RFM12B support from (running on Atmega 328) is disabling the interrupt in the interrupt mask register when entering the ISR anyway so this should also work. I need to test if `pio.pin.IntrLowLevel` could be better than `pio.pin.IntrNegEdge` in my case. The RFM12B module keeps the nIRQ low until it gets a proper command, depending on the interrupt reason.

jolivepetrus, 2018-05-03:
Hi @yawor,

Issue with pio.pin.IntrLowLevel is solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/439261cccebb3b8daa4d631a53d0352546660139.

yawor, 2018-05-03:
@jolivepetrus 

I see what you did in that commit. It doesn't starve the CPU anymore, but I don't think this is an expected behaviour. Right now you're emulating IntrNegEdge and IntrPosEdge behaviour for IntrLowLevel and IntrHighLevel respectively so there's practically no real difference between them (besides being less optimal).
I think the expected behaviour for IntrLowLevel and IntrHighLevel is to re-queue the callback as long, as the state of the pin has selected value. So when the callback finishes, the C code calling the Lua callback function should check the pin state and re-queue itself if needed. The tricky part is probably to do this in a way so it won't starve the CPU again.

jolivepetrus, 2018-05-03:
@yawor,

I'm a bit confusing about the behaviour of a GPIO interrupt of type GPIO_INTR_HIGH_LEVEL / GPIO_INTR_LOW_LEVEL. In my tests, using only esp-idf, the interrupt is fired indefinitely until the gpio logic level changes.

yawor, 2018-05-03:
@jolivepetrus 

Yeah, it should work like that, because that type of interrupt is not triggered by the edge (pin state change), but by a pin state itself. So for IntrLowLevel, the callback should be re-queued after it finishes as long, as the pin is pulled low. Now how to do it without starving the CPU - I don't know.
With your last commit, after triggering the callback, the pin still needs to go high and then low again to fire it again, so it behaves like IntrNegEdge.

jolivepetrus, 2018-05-03:
@yawor,

I have reverted the commit, and I see that in my case the CPU was starved due to that I attach the interrupt into a very high priority:

```
-- This starves
pio.pin.interrupt(pio.GPIO14, function(value)
...
...
end, pio.pin.IntrLowLevel, nil, nil, 25)
```

```
-- This don't starves
pio.pin.interrupt(pio.GPIO14, function(value)
...
...
end, pio.pin.IntrLowLevel, nil, nil, 20)
```

FYI the Lua interpreter has a priority of 20.

jolivepetrus, 2018-05-09:
@yawor, @xopxe,

Please, discuss the performance issues in https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/162.

jolivepetrus, 2018-05-29:
@yawor,

About NSS timing, please try with commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/05bd970013556eb8a0b06efe8594495a772a2bdf, it should work now.

the0ne, 2019-04-09:
@yawor please comment or close this issue - thanks!

yawor, 2019-04-09:
I'm sorry for the lack of input from me. I've moved on with working directly with esp-idf and platformio/esphome. I think there'll always be too much overhead with LuaRTOS regarding SPI for my specific use case. I still think that LuaRTOS is great piece of software but it's just not what I need for this project.

closed 2019-04-09 by yawor