whitecatboard/Lua-RTOS-ESP32 #153: GPS Sensor Error: "invalid data bits"

mafrmt00, 2018-04-26:
On "SparkFun ESP32 Thing"
When using the GPS Sensor with:
gps = sensor.attach("GPS", uart.UART2, 9600, 8, uart.PARNONE, uart.STOP1)

It always fails with the error:
/_run.lua:1: 67108866:invalid data bits 
stack traceback:  
[C]: in field 'attach'  
/_run.lua:1: in main chunk  
[C]: in global 'dofile'  
stdin:1: in main chunk  
[C]: in ? 

The UART2 shows the NMEA Data without problems, with the Terminal Example: https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/components/spiffs_image/default/examples/lua/uart-terminal.lua

yawor, 2018-04-26:
I've checked the source code and from what I can see only UART id (uart.UART2) is passed to the sensor driver.
In `components/lua/modules/hw/sensor.c` file there's a function called `lsensor_setup_prepare`. It checks the type of the sensor and retrieve arguments depending on that type. As you can see here
```lua
                                case UART_INTERFACE:
                                        setup[i].uart.id = luaL_checkinteger(L, top++);
                                        break;
```
only the UART id is taken from the list of arguments despite the `sensor_setup_t.uart` structure having all the other needed fields. They are left initialised to 0 and never set. That's why the error.

jolivepetrus, 2018-04-29:
@mafrmt00, @yawor,

This is solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/a0ef83a6053605db08cc618bbb4a9d184bd3d44b.

yawor, 2018-05-01:
@jolivepetrus I'm wondering if the current way of providing arguments to configure sensors is optimal. For example, when using it this way, when a sensor uses multiple interfaces, then the optional arguments for a specific interface type can get messy.
I've been thinking if it could be made better by using tables, for example instead of:
```lua
sensor.attach("ANALOG_JOYSTICK", adc.ADS1115, 0, adc.ADS1115, 1, pio.GPIO26)
```
it could look like
```lua
-- sensor.attach(id, args_for_if1, args_for_if2, args_for_if3, ...)
sensor.attach("ANALOG_JOYSTICK", {adc.ADS1115, 0}, {adc.ADS1115, 1}, {pio.GPIO26})
```
That way you don't need to worry that adding some extra arguments for specific interface could mess with the order of arguments for other interfaces. For example, if you would add an optional argument to ADC, like prescaller or resolution options, then it won't break the code which doesn't use these args.
Optionally, the keys in the arguments table could be descriptive (`{source=adc.ADS1115, channel=0}`) or even mixed (required args could be positional, but optional would use a descriptive key like `{adc.ADS1115, 0, precision=10}`).
Second optional idea could be for interfaces, where there's only one required argument, you could omit the table and give the argument directly to `sensor.attach()`.

What do you think about such change? I know it may be hard to do something like this without breaking code which uses current API. But it the long run, it may be worth doing it.

closed 2018-05-04 by mafrmt00