whitecatboard/Lua-RTOS-ESP32 #157: CJSON: doesn't encode values correct

lu1210, 2018-04-28:
tbl = {}
tbl.value = 1234475220
print(tbl.value)
require('cjson').encode(tbl)

/ > tbl = {}
/ > tbl.value = 1234475220
/ > print(tbl.value)
1234475220
/ > require('cjson').encode(tbl)
{"value":1234475264}
/ >

jolivepetrus, 2018-04-29:
@lu1210,

In early versions of Lua there was not differentiation between integers and floats, so integers where represented as float in that versions.

Lua RTOS is based in the last Lua version (5.3.4), that has differentiation between integers and floats. As far I can see, the current lua_cjson version treats integers as floats.

The commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/fcce82c2b279a586fb120e273b61f56d50d24c19 adds integer / float differentiation in code and decode. Please try again.

lu1210, 2018-04-30:
hi, I have checked 3 test cases

a) number is an INTEGER    less equal 3 byte.
   - encoding with cjson is ok
   - INTEGER view with string,format is ok

/ > tn = tonumber
/ > fm = string.format
/ > tbl = {}
/ > tbl.value = 16777216  -- 0v01000000
/ > print(tbl.value)
16777216                  
/ > require('cjson').encode(tbl)
{"value":16777216}             <--------------- ok
/ > print(string.format('%.14g',tbl.value))
16777216                       <----------------ok

---------------------------------------------------
b) number is an INTEGER  greater 3 byte.
   - encoding with cjson is ok
   - INTEGER view with string,format is **not ok**
   

 > tn = tonumber
/ > fm = string.format
/ > tbl = {}
/ > tbl.value = 16777216 +1 -- 0v01000000 +1
/ > print(tbl.value)
16777217
/ > require('cjson').encode(tbl)
{"value":16777217}             <--------------- ok
/ > print(string.format('%.14g',tbl.value))
16777216                       <--------------- ## not ok
/ > 
--------------------------------------------------

c) number is a REAL greater 3 byte.
   - encoding with cjson is not ok
   - INTEGER view with string,format is not ok
   
/ > tn = tonumber
/ > fm = string.format
/ > tbl = {}
/ > tbl.value = 16777216 +1.0  -- 0v01000000 +1
/ > print(tbl.value)
1.677722e+07
/ > require('cjson').encode(tbl)
{"value":16777216}             <--------------- not ok
/ > print(string.format('%.14g',tbl.value))
16777216

jolivepetrus, 2018-05-03:
@lu1210,

My god! I hate to work with real numbers!

As far I can see, this issue is caused by the fact that in Lua RTOS real numbers use 32-bit floats.

To solve this, now Lua RTOS can be configured to work with 64-bit integer / real numbers. In my tests, all the above test cases work as expected.

See commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/37c9ce090b00ffece6809e0d015cd203308c6ef9.

lu1210, 2018-05-04:
64 bits INTEGER are great for ESP32:-) It works fine, but there could be some side affects in drivers like this one 

  tds1820 = sensor.enumerate(sensor.OWire, pio.GPIO4,true)
/ >    require('show')(tds1820)
  1.
    id = 'DS1820'
    addressl = 1897202083
    addressh = 687853426
    model = 'DS18B20'
    device = 2
  0.
    id = 'DS1820'
    addressl = -2146172164   <---- number is negative
    addressh = 687819334
    model = 'DS18B20'
    device = 1

jolivepetrus, 2018-05-04:
Yes, I know. For this reason the feature is marked as experimental, and not enabled by default.

+1 side affect

lu1210, 2018-05-09:
is a  32-bit integers and double-precision (64-bit) floats configuration possible?

closed 2018-06-20 by lu1210