whitecatboard/Lua-RTOS-ESP32 #158: GPS Position is off by about 1km

mafrmt00, 2018-05-01:
When using the GPS Sensor, the position is off about 1.2km to the south. 
The position rarely changes, even the last digit ist very stable.
When the receiver is moved indoors, the position goes to 0, also the satellite count.

When the NMEA Data of the GPS is dumped with the Terminal example and analyzed, the position is correct.

mafrmt00, 2018-05-06:
In Lua-RTOS-ESP32/components/nmea/nmea0183.c

Line 122:            
seconds = (double)atoi(current) / (double)1000000;

Should be:
seconds = (double)atoi(current) / (double)10000;

gmenier, 2018-07-10:
Same problem here. There is indeed an offset (NEO-6M).

the0ne, 2019-04-09:
@mafrmt00 can you create a [PR](https://github.com/whitecatboard/Lua-RTOS-ESP32/pulls) for this?

adrianvielsack, 2019-07-01:
#277 should fix this, regardless of precision

the0ne, 2019-07-08:
@adrianvielsack Thanks a lot!
@jolivepetrus what do you think about #277?

the0ne, 2019-07-31:
@adrianvielsack thanks for your fix - it has been accepted into Lua-RTOS!
@jolivepetrus thanks for merging!

closed 2019-07-31 by the0ne