whitecatboard/Lua-RTOS-ESP32 #159: SSH Connection Fails

mafrmt00, 2018-05-02:
SSH connection is not possible. Neither from Linux, MacOS or Windows.

ESP32 Module Log:
dropbear: Child connection from 192.168.6.94:51323

Windows Bitvise SSH Client Log:
09:53:47.993 Started a new SSH2 session.
09:53:48.012 Connecting to SSH2 server 192.168.6.193:22.
09:53:48.215 Connection established.
09:53:48.240 Server version: SSH-2.0-dropbear_2017.75
09:53:48.240 First key exchange started.
09:53:48.416 Received host key from the server. Algorithm: RSA, size: 2048 bits, MD5 fingerprint: 60:93:29:e8:28:8e:c7:b7:12:a3:5f:b7:22:45:c6:19, Bubble-Babble: xivem-myzyn-vaber-gyvih-zypag-tidim-torif-lonoh-ferac-rafel-luxex, SHA-256 fingerprint: babWff86pOflRNewz3G7OE3fsFLg2BHin6WLZTLxVU4.
09:53:48.463 The SSH2 session has terminated with error. Reason: Error class: LocalSshDisconn, code: KeyExchangeFailed, message: FlowSshKexDhClient: invalid DH value.

jolivepetrus, 2018-05-03:
@mafrmt00,

I can connect to Lua RTOS from windows with the bitwise ssh client:

[image]

My log:

12:30:29.368 Current date: 2018-05-03
12:30:29.368 Bitvise SSH Client 7.41, a fully featured SSH client for Windows.
Copyright (C) 2000-2018 by Bitvise Limited.
12:30:29.368 Visit www.bitvise.com for latest information about our SSH software.
12:30:29.368 Run 'BvSsh -help' to learn about supported command-line parameters.
12:30:29.368 Cryptographic provider: Windows CNG (x86) with additions
12:30:29.549 Loading default profile.
12:30:29.549 Loading default profile failed: RegOpenKeyExW() failed: Windows error 2: El sistema no puede encontrar el archivo especificado.
12:30:29.549 Loading a blank profile.
12:30:51.778 Started a new SSH2 session.
12:30:51.784 Connecting to SSH2 server 192.168.1.46:22.
12:30:51.951 Connection established.
12:30:51.959 Server version: SSH-2.0-dropbear_2017.75
12:30:51.959 First key exchange started. Cryptographic provider: Windows CNG (x86) with additions
12:31:00.833 Received host key from the server. Algorithm: RSA, size: 2048 bits, MD5 fingerprint: 4b:30:d1:1c:ac:a8:05:f1:5d:61:7e:bf:b4:e5:37:7f, Bubble-Babble: xuriz-tylob-mizap-detyh-mitof-kuzet-hamyc-bilom-ladat-bonin-lexex, SHA-256 fingerprint: KVrvJlZnebuxKnhxsr08RE3jyKPyIUu0K8bQ44FRyPk.
12:31:05.329 Host key has been saved to the global database.
12:31:05.351 First key exchange completed using diffie-hellman-group14-sha1. Session encryption: aes256-ctr, integrity: hmac-sha2-256, compression: none.
12:31:05.467 Attempting none authentication.
12:31:05.812 Remaining authentication methods: 'publickey,password'.
12:31:10.846 Attempting password authentication.
12:31:11.007 Authentication completed.
12:31:11.018 Terminal channel opened.
12:31:11.018 SFTP channel opened.
12:31:11.024 SFTP channel closed. SFTP request failed.

yawor, 2018-05-03:
I'm also having a problem connecting using OpenSSH on Arch Linux. I'm getting:
`ssh_dispatch_run_fatal: Connection to 192.168.88.121 port 22: error in libcrypto`

The same happens from a router running Debian Stretch.

This is strange, because the first time I've installed Lua-RTOS on my ESP32 board, I've played with WIFI and SSH and it worked properly. I then started with a fresh spiffs file system without enabling WIFI and SSH because I don't need them at this moment. Now I enabled them, waited for rsa and dss keys to be generated and now I'm getting this error. The root password is set.

mafrmt00, 2018-05-04:
It seems to be that a invalid configuration is not detected. I was able to connect via ssh, but only after I erased the board and flashed it new.
But the key generation started only after a reset. The first net.service.ssh.start() did not work.

My Steps were:
wcc -p COM3 erase
wcc -p COM3 -f -ffs

LUA: passwd
LUA: net.service.ssh.start() <- no success

Board: reset
LUA: net.service.ssh.start() <- key generation and successful connection

yawor, 2018-05-04:
@mafrmt00 you're right. It seems to be some kind of fs issue. I've erased the flash and started fresh and SSH now works.

spellauge, 2019-01-15:
i tried all of the above but still i get 

ssh_dispatch_run_fatal: Connection to 192.168.2.10 port 22: error in libcrypto

when trying to connect to my esp32 using ssh.

my client's linux environment:

Distributor ID: Debian
Description:    Debian GNU/Linux buster/sid
Release:        testing
Codename:       buster

ii  openssh-client                                1:7.9p1-4                            amd64        secure shell (SSH) client, for secure access to remote machines

at least at my site ssh-server on esp32 seems to be unusable.

the0ne, 2019-04-09:
@mafrmt00, @yawor and @spellauge please enable outputting all log using
`os.loglevel(os.LOG_ALL)`
before starting the ssh server on the esp32.
Then, please share the output.

What I could imagine is that the esp32 is rebooted - e.g. accidentally or by crash - while dropbear (which is the ssh-server compontent) is still generating it's key files during it's initial run.
In that case you would have to delete the files using
`rm /etc/dropbear/dss_host_key` and `rm /etc/dropbear/rsa_host_key` and then restart the ssh-server.
Then wait until dropbear has finished generating it's key files before unpowering/crashing/resetting the board.

the0ne, 2019-07-18:
@mafrmt00, @yawor and @elsbiet please provide feedback - thanks!

the0ne, 2019-07-26:
@mafrmt00, @yawor and @elsbiet please provide feedback - thanks!
Closing this issue until then.

closed 2019-07-26 by the0ne