whitecatboard/Lua-RTOS-ESP32 #160: MQTT client connection

gmenier, 2018-05-02:
mqtt.client(...) should wait a little more (or a timeout anywhere is a bit short)

`client = mqtt.client("clientId-012345", "mylittlebroker.com", 1883, false)

tmr.delay(2) -- <<<< not a solution but it helps :-(

client:connect("","")
client:publish("testtopic","hola",mqtt.QOS0) 
client:disconnect()`

Tested with a local mosquitto and most of the free online mqtt brokers : the lua code sometimes aborts before the connection with the broker (and then of course, client:connect() fails). 

A messy delay helps (?).

Regards
Gildas

jolivepetrus, 2018-05-04:
@gmenier,

I have increased the connection timeout from 10 secs to 20 secs.

Please, try with https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/d418fcb814d5d7e8743cd387df75790b913c8aed.

gmenier, 2018-05-05:
I still have the problem. It may not be a timeout problem after all - perhaps a blocking call that doesn't block. 
Sometimes, the mqtt.client call doesn't wait for a connection and ends immediately with no error (not waiting 10 or 20s - broker's side the -verbose log shows that no attempt to contact the broker has been made). the mqtt.client raises no exception. It just ends and lua runs the connect call.

The following call of client:connect raises immediately an exception :  
lua: /_run.lua:4 201326594:can't connect
because no connection is available.

It seems that mqtt.client should wait till a connection is made or a timeout. Sometimes the connection is fast and happens before the client:connect, sometimes not (?). A delay helps there.

I know the 'sometimes it fails' issue is not easy to fix. :-(

Regards

Gildas

jolivepetrus, 2018-05-23:
@gmenier,

Can you try with newer Lua RTOS versions, I have published some changes related to MQTT.

gmenier, 2018-05-30:
much better :-) Thank you !
Great !!
G.

closed 2018-05-30 by gmenier