whitecatboard/Lua-RTOS-ESP32 #162: JIT bytecode optimizer

jolivepetrus, 2018-05-09:
As Lua RTOS makes intensive use of read-only tables, a series of optimizations can be performed on the program's bytecode to speed-up the program execution. This can help that programs written for Lua RTOS to have a similar performance than the writtens in C, and takes a special importance when the programmer use the Lua RTOS hardware-access modules.

We have start the work on an initial version of the Lua RTOS JIT bytecode optimizer, with excellent results.

jolivepetrus, 2018-05-09:
I have some initial experimental results.

Objective: know how many time is needed by Lua to respond to a GPIO interrupt.

Test code:

```
pio.pin.setdir(pio.OUTPUT, pio.GPIO21)
pio.pin.interrupt(pio.GPIO14, function(value)
    pio.pin.sethigh(pio.GPIO21)
    tmr.delayus(1)
    pio.pin.setlow(pio.GPIO21)
end, pio.pin.IntrRaiseEdge)

device = pwm.attach(pio.GPIO26, 1000, 0.1)
device:start()
```

Wire: GPIO26 to GPIO14
Test points: GPIO14 and GPIO21

CASE A: Lua RTOS with Lua locks & without JIT bytecode optimizer

[image]

CASE B: Lua RTOS with Lua locks & JIT bytecode optimizer

[image]

CASE C: Lua RTOS without Lua locks & JIT bytecode optimizer

[image]

Conclussions:

CASE A: use C if you need a response time < 60 usecs
CASE C: use C if you need a response time < 35 usecs
CASE D: use C if you need a response time < 16 usecs

It is clear that the JIT has a good impact in the performance, and it is mandatory to determine in which situations it is safe to disable Lua locks.

jolivepetrus, 2018-05-11:
Hi guys,

A first version of the JIT byte-code optimizer is available in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/6763f11fa9a2d3c8b615db6905a5663f642f4254.

yawor, 2018-05-11:
@jolivepetrus what exactly are Lua locks? When it's safe to disable them? Is it needed only when multiple threads are accessing the same hardware? If I use only a dedicated thread to access a specific hardware or I use explicit mutexes in my code to disallow simultaneous access to it, can I safely disable the Lua locks?

jolivepetrus, 2018-05-11:
@yawor,

Lua locks are recursive mutexes that are used in the lua API to protect concurrent access to the Lua state. For example, when calling the lua_newuserdata, or the lua_pushinteger function, a lock is adquired before modifying the lua state (the structure that holds, for example the global variables) and released when the lua state is modified.

The thing is that Lua RTOS is programmed in a way that lua locks can be omitted in certain circumstances:

* The callbacks use a new lua state (in fact it is a light version of the lua_State structure). The code executed in the callback don't require to use locks when accessing the lua api.

* Lua RTOS threads also use a new lua state, so the above rule can be applied to it.

* Within a callback, or a Lua RTOS thread, only the lua api functions that access to global variables must be protected by a lua lock.

With the actual Lua RTOS version, it should be safe to disable locks if:

* To run programs, put the source code into a lua file, and use dofile to execute the program. Avoid paste code in the console.

* Always protect access to global variables from Lua RTOS threads (use the mutual exclusion functions from the thread module). 

As far I can see, disabling lua locks, or minimize the use of lua locks it is feasible.

yawor, 2018-05-11:
@jolivepetrus thanks for the explanation
Unfortunately enabling JIT with disabled Lua Locks at the same time has a negative impact on SPI.

I've been testing this by executing 10-12 consecutive spi:readwrite operations, each sending 2 bytes. Nothing more in the program right now (wifi and network services disabled). I recorded the time between the NSS going down and back again and the time to NSS going down again.
- LL on, JIT off: 260 us, 46 us
- LL off, JIT off: 140 us, 22 us
- LL on, JIT on: 180 us, 24 us
- LL off, JIT on: 180 us, 25 us
So I got the best performance with both LL and JIT off. It's strange that with JIT on, there's almost no difference on whether the LL is on or off.

BTW is disabling hardware locks supported? I wanted to try disabling them in Kconfig, but then I got some error with i2c unlock function missing during compilation.

jolivepetrus, 2018-05-11:
@yawor,

Please, attach your test code, to check the JIT optimizations.

jolivepetrus, 2018-05-11:
@yawor,

In theory the JIT's commits disables lua locks and read-only table cache if JIT is enabled:

Cache:

When accessing to readonly tables, Lua RTOS can get the key/value pair from a cache. This can
speedud the execution of Lua RTOS scripts. This option is disabled when the JIT bytecode
optimizer is enabled.

Lua locks:

Use locks when the program enters the Lua core. This option is disabled when the JIT bytecode
optimizer is enabled.

For this reason you have no differences with JIT=on & (LL=on | LL = off). If JIT=on LL are always disabled.

Now LL setting is present in Kconfig for compatibility, and will be removed from Kconfig soon.

yawor, 2018-05-11:
Ok, here's a little twist. I have two programs: one that I started implementing communication with RFM12B module and a stripped down which only enables SPI and sends some bytes.
I've actually tested this on the bigger program and now I've checked on the stripped down one. What's strange, the stripped down version is slower than the full one (on compilation where JIT is enabled, I get 200-210 us instead of 180 us like I've posted earlier). The SPI code is almost identical (besides function name) and the difference is that I start the thread before running init commands over SPI in the full program.

Here's stripped down code:
```lua
local spiData = {
   0x0123,
   0x4567,
   0x89ab,
   0xcdef,
   0x0000,
   0x1111,
   0x2222,
   0x3333,
   0x4444,
   0x5555,
   0x6666,
}

local dev

local function readwrite(data)
    dev:select()
    local ret = dev:readwrite((data & 0xff00) >> 8, data & 0xff)
    dev:deselect()
    return ret[0] << 8 | ret[1]
end

function runtest()
    dev = spi.attach(spi.SPI3, spi.MASTER, pio.GPIO5, 2000000, 8, 0)
    for i, cmd in ipairs(spiData) do
        readwrite(cmd)
    end
end
```

And here's the full one:
```lua
-- RFM12B

local FREQUENCY = 1664 -- math.floor((868320000 - 860000000) / 5000) -- 868.320 MHz
local FSK_SHIFT = 48 -- (math.floor(60000 / 15000) - 1) << 4 -- 60000
local DATARATE = 35 -- math.floor((10000000 / 29 / 9600) - 0.5)

local CMD_CFG = 0x8000
local CFG_EL = 0x80
local CFG_EF = 0x40
--local CFG_BAND_315 = 0x00
--local CFG_BAND_433 = 0x10
local CFG_BAND_868 = 0x20
--local CFG_BAND_915 = 0x30
--local CFG_XTAL_8_5PF  = 0x00
--local CFG_XTAL_9_0PF  = 0x01
--local CFG_XTAL_9_5PF  = 0x02
--local CFG_XTAL_10_0PF = 0x03
--local CFG_XTAL_10_5PF = 0x04
--local CFG_XTAL_11_0PF = 0x05
--local CFG_XTAL_11_5PF = 0x06
--local CFG_XTAL_12_0PF = 0x07
local CFG_XTAL_12_5PF = 0x08
--local CFG_XTAL_13_0PF = 0x09
--local CFG_XTAL_13_5PF = 0x0A
--local CFG_XTAL_14_0PF = 0x0B
--local CFG_XTAL_14_5PF = 0x0C
--local CFG_XTAL_15_0PF = 0x0D
--local CFG_XTAL_15_5PF = 0x0E
--local CFG_XTAL_16_0PF = 0x0F

local CMD_PWRMGT = 0x8200
local PWRMGT_ER = 0x80
[... 179 lines cut ...]
end

function rfm12_callback()
    while true do
        status = rfm12_readwrite(CMD_STATUS)
        recheck = false
        if status & STATUS_FFIT ~= 0 then
            recheck = true
            recvd = recvd + 1
            buffer[recvd] = rfm12_readwrite(CMD_READ) & 0xff
            if recvd == 11 then
                rfm12_readwrite(CMD_CLEAR_FIFO)
                rfm12_readwrite(CMD_ACCEPT_DATA)
                table.insert(packets, buffer)
                buffer = {}
                recvd = 0
                packet_rcvd:broadcast()
            end
        end
        if not recheck then break end
    end
end
```

Don't worry about the rest of the program in the full code. I'm testing it right now without anything connected to the SPI except the logic analyser. When you run `rfm12_init` function, it should send about 13 transactions, each containing 2 bytes.

jolivepetrus, 2018-05-15:
@yawor,

New optimizations have been added in the JIT (see https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/f88aa17ec250b54acf1259d8d937ada54e91b2e3).

Also there are changes in the Lua spi module (readwrite function). Now the function returns an array t[k] with k >= 1, that ensures that the result table is an array and no table resizes are done when the table is build. In the previous version the array was t[k], with k >= 0, but this was not an array (in Lua arrays are indexed from 1 to array size) and caused continuous table resizes.

You should see a 150 usecs time between NSS with the JIT enabled. It is not feasible to get lower values without changes in the SPI driver, but I have some ideas to implement in the driver to reduce the transaction duration.

jolivepetrus, 2018-05-29:
@yawor,

Please try with commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/05bd970013556eb8a0b06efe8594495a772a2bdf. Now the SPI Lua module uses an internal buffer to make transfers to the SPI bus.

yawor, 2018-05-31:
@jolivepetrus 
I can confirm that the SPI operation now looks a lot better. I don't have the RF module connected right now (just logic analyser), so I don't know if it's now enough to properly handle the transmission. I'll check it during the weekend.

yawor, 2018-06-03:
@jolivepetrus 
First of all, thank you for your great support regarding this issue.

I don't know if this is a good issue to post this, maybe I should open a new issue for the SPI?
Unfortunately by making recent you've introduced an error in the spi:readwrite function. It now returns the same data as given in the arguments, instead of the real, received data. I've confirmed that it's not the slave device causing this with the Logic Analyser and the same thing happens when I disconnect the MISO pin (it should return zeros when MISO is not connected).

I've tested the SPI performance and my conclusion is that the moment the NSS pin is activated/deactivated only helps partially. I've tried with and without the new `SPI_FLAG_CS_AUTO` to time both approaches and the total time to send 10 SPI requests is almost the same. When I test the time between last falling clock edge of the previous transfer to the first rising clock edge of the next transfer, it is about 190 us with or without the flag. I don't know if this time can be shortened further - probably not much. It can certainly go up: just enabling wifi in the config.lua adds ~20 us to this time.

BTW I've noticed that the NSS now rises up at the exact same time as the last clock edge falls down on the transfer. Is it possible to introduce a little delay? The time between NSS going down and the first clock edge is 1/2 of clock cycle, which should also be ok for the NSS going up after the last clock edge. Or add ability to add a number of dummy bits at the end. Some devices don't like NSS going up on the last clock edge. For example, the RFM12B datasheet calls this time "Select hold time" and requires it to be at least 25 ns.

jolivepetrus, 2018-06-04:
We can introduce a small delay before NSS goes high, but not when NSS goes low due to en ESP32 errata (this only works in half-duplex). Maybe it's better to make SPI_FLAG_CS_AUTO = 1 by default, this introduces a small delay.

jolivepetrus, 2018-06-06:
@yawor,

Reading issue is solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/ba7d71b5efac131a37d13c9b0d94c97120fc0771.