whitecatboard/Lua-RTOS-ESP32 #163: Sensor timeout when wifi on (sta)

gmenier, 2018-05-09:
s = sensor.attach("DHT11", pio.GPIO15)

while true do
  print("temp: "..s:read("temperature").." humidite : "..s:read("humidity"))
  tmr.delayms(500)
end

fails (after a few iterations) to a timeout in the method read when wifi (sta) is on (started thru the config.lua). 
 
Works perfectly when wifi is off (net.wf.stop()).

Tested using 5 different DHT11. EspThing latest build.

Regards
G.

jolivepetrus, 2018-05-15:
@gmenier,

I can't reproduce with all upgraded.

Please, from Lua RTOS root folder, try:

git pull
make upgrade-idf
make flash

... but DHT sensor series must be implemented using the RMT module for best performance (see https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/165).

jolivepetrus, 2018-05-25:
@gmenier,

This is solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/9955a645464a81e44f14e4e5c62e2983039bfc40.

gmenier, 2018-05-25:
Nice ! Thank you (amazing support :-) ). Closing.

closed 2018-05-25 by gmenier