whitecatboard/Lua-RTOS-ESP32 #166: Support for time-of-flight distance sensors: VL53L0X

xopxe, 2018-05-15:
We need support for the VL53L0X distance sensors. There is functional code for Arduino here: https://github.com/pololu/vl53l0x-arduino .
If you do not have time or interest in adding it we're ready to put work on this, but it would be great to get some guidance.

xopxe, 2018-05-22:
Uh, sorry for the noise, the question 2 was just typo when defining flags. Going forward.

jolivepetrus, 2018-05-23:
@xopxe,

Nice to hear your interest in the VL53L0X driver for Lua RTOS. Sorry for the delay in my answer, so I was involved in other Lua RTOS internals.

Please, send us any required information to give support.

xopxe, 2018-05-23:
Thanks for the support. We need several sensors, which have a hardcoded i2c address. You can change the number, but it will reset on power cycling. On boot you are supposed to power them up one by one, changing the i2c address. How would you do that with the "sensors" infrastructure? Add a "adress" attribute? Would that work?
We are needing a distance "sensor ring", perhaps there's another approach? Perhaps a single "virtual sensor" from which you get all the reading in a single array? Of course, the single "sensor" is more generally useful...

jolivepetrus, 2018-05-23:
@xopxe,

Extracted from our wiki (https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Sensor-module#configuration-functions) / attach function:

_For I2C:

id: a string containing the sensor id, for example BME280.
i2c: i2c unit. Use i2c.I2Cx defined for this purpose.
address: i2c device address. If 0 the default device address is applied._

In the sensor implementation you have to deal with this. Extracted from the bme280:

driver_error_t *bme280_presetup(sensor_instance_t *unit) {
	// Set default values, if not provided
	if (unit->setup[0].i2c.devid == 0) {
		unit->setup[0].i2c.devid = BME280_I2C_ADDRESS1;
	}

	if (unit->setup[0].i2c.speed == 0) {
		unit->setup[0].i2c.speed = 400000;
	}

	return NULL;
}

In the above code, unit->setup[0] is the reference of the first interface setup that is created in the sensor lua api, then you access to the i2c configuration (unit->setup[0]), in which you have the arguments of the attach function. For example:

s = sensor.attach("BME280", i2c.I2C0, 0):

      unit->setup[0].i2c.devid = 0

s = sensor.attach("BME280", i2c.I2C0, 0x30):

      unit->setup[0].i2c.devid = 0x30

xopxe, 2018-05-23:
So, if I add a "set" to change the i2c address I would have to update unit->setup[0].i2c.devid, and then it would work normally? 
Also I need a digital GPIO per sensor to enable/disable them while renumbering. Should I passs it to the device as an attribute? something like that:
```
s = sensor.attach("VL53L0X", i2c.I2C0, 0x30)
s:set("enablepin", pio.GPIO19)
//later on
s:set("enable", true)
```

jolivepetrus, 2018-05-23:
No, no, the i2c address is used when the sensor is attached. During the live of the sensor instance, the i2c address never changes.

What is the use-case for the GPIO you need? It's a GPIO connected to the sensor?

xopxe, 2018-05-24:
The problem with these sensors is that they have a hardcoded i2c address. It can be changed trough a i2c write, but the change will not survive a reboot. There are the docs on how to use several on them on the same bus: www.st.com/resource/en/application_note/dm00280486.pdf
Without extra hardware the only way to use them is to renumber them on boot:
1) have the reset pins (XSHUT) from each sensor connected to the gpio.
2) put all devices in reset
3) for each device, take it out of reset, and use the hardcoded i2c address to do a write setting a new i2c address

Perhaps the best for us is to have a separated module, not integrated with the sensor support?

xopxe, 2018-05-25:
I see what you mean now (i think), I was mixing i2c unit and device addresses...
Still have to do the renumbering dance, I think I see a way forward.

closed 2018-05-25 by xopxe