whitecatboard/Lua-RTOS-ESP32 #168: Http server

gjongenelen, 2018-05-23:
As stated in #24 the http-server will be transferred to the new luasocket. 

Is this done already? Because I get panic when printing via lua. 

For example:
`
print("Hello")
`
I get a kernel panic.

jolivepetrus, 2018-05-23:
Hi @gjongenelen,

The related issue is a very old issue. Luasocket is available in Lua RTOS since commit https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/c318dc8853fdd12df992b9e51b504f8afe854795. In this moment we have no plans to implement a webserver using the luasocket library, because we prefer the current C implementation.

Please, provide example lua sripts that fires the kernel panic you mentioned above.

gjongenelen, 2018-05-23:
Well, the following content in autorun.lua:
```
os.logcons(false)
os.shell(true)

net.wf.setup(net.wf.mode.STA, "ssid","password)
net.wf.start()
net.service.sntp.start()

net.service.http.start()
```
and in the index.lua:
```
print("hello")
```
Triggers (when requesting http://192.168.0.80): 
``` 
Guru Meditation Error: Core 1 panic'ed (LoadProhibited)
```

gjongenelen, 2018-05-29:
Here you go: 
```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V=
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y

#
# Security features
#
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/tty.SLAB_USBtoUART"
CONFIG_ESPTOOLPY_BAUD_115200B=y
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=y
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHFREQ_26M=
CONFIG_ESPTOOLPY_FLASHFREQ_20M=
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
CONFIG_ESPTOOLPY_FLASHSIZE_1MB=
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
CONFIG_ESPTOOLPY_BEFORE_NORESET=
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_RESET=y
CONFIG_ESPTOOLPY_AFTER_NORESET=
CONFIG_ESPTOOLPY_AFTER="hard_reset"
CONFIG_MONITOR_BAUD_9600B=
CONFIG_MONITOR_BAUD_57600B=
[... 741 lines cut ...]
CONFIG_LUA_RTOS_LUA_USE_PWM=y
CONFIG_LUA_RTOS_LUA_USE_SENSOR=y
CONFIG_LUA_RTOS_LUA_USE_SERVO=y
CONFIG_LUA_RTOS_LUA_USE_SOCKET=y
CONFIG_LUA_RTOS_LUA_USE_SPI=y
CONFIG_LUA_RTOS_LUA_USE_SCP_NET=y
CONFIG_LUA_RTOS_LUA_USE_SDISPLAY=
CONFIG_LUA_RTOS_LUA_USE_STEPPER=y
CONFIG_LUA_RTOS_LUA_USE_THREAD=y
CONFIG_LUA_RTOS_LUA_USE_TMR=y
CONFIG_LUA_RTOS_LUA_USE_UART=y
CONFIG_LUA_RTOS_LUA_USE_ULP=y

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
```

jolivepetrus, 2018-06-08:
@gjongenelen,

Founded, and solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/b6214a3912eadc5d40772522e57abd1ab64c2216. Please, reopen the issue if you need more assistance.

closed 2018-06-08 by jolivepetrus