whitecatboard/Lua-RTOS-ESP32 #178: Captive Web server

pocampov, 2018-06-11:
I appreciate a help. I try to make a captive web server in my ESP-32 board. I used this lines:
```
net.wf.setup(net.wf.mode.AP,"Finca","")
net.wf.start()
-- Sube el servidor Web en el puerto 80
net.service.http.start()
net.stat()

-- Activa DNS cautivo
net.service.captivedns.start()
```

but i get this error: network is not available

the0ne, 2018-06-19:
try to replace
`net.wf.start()`
by
`net.wf.start(false)`

jolivepetrus, 2018-06-19:
@pocampov, @the0ne,

It should be solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/bf4e0b705a7ef9b7731a750181eeb667791bc7e5.

No matters if using net.wf.start(false) or net.wf.start(true) now.

Please, reopen it the issue persists.

closed 2018-06-19 by jolivepetrus

pocampov, 2018-07-06:
@Jolivepetrus and @the0ne thanks for your help. 
I reopen #178 because the problem persists.
My version is Lua RTOS	beta 0.1	1528456632
the exact error is:
`lua: /system.lua:9: 150994944:network is not available `

The code is:
`
net.wf.setup(net.wf.mode.AP,"Finca","",
    net.wf.powersave.NONE, 4, false)
net.wf.start(true)

net.service.http.start()
net.stat()

-- Activa DNS multicast
net.service.captivedns.start()
`

I thank you in advance for your collaboration