whitecatboard/Lua-RTOS-ESP32 #183: OSTICKS_PER_SEC

gmsotavio, 2018-07-11:
Hi,

I've noticed that in lmic_hal.c the timer was adjusted in such way that OSTICKS_PER_SEC = 50000 that seems ok. However in oslmic.h the current definition is OSTICKS_PER_SEC = 32768. Have you sure that is it right?

the0ne, 2019-04-09:
@gmsotavio you are right in the fact that those defines don't match. But if you look closely, the code is as follows:
```
#ifndef OSTICKS_PER_SEC
#define OSTICKS_PER_SEC 32768
#elif OSTICKS_PER_SEC < 10000 || OSTICKS_PER_SEC > 64516
#error Illegal OSTICKS_PER_SEC - must be in range [10000:64516]. One tick must be 15.5us .. 100us long.
#endif
```

That first line `#ifndef OSTICKS_PER_SEC` makes sure that this `#define OSTICKS_PER_SEC 32768` is only done if OSTICKS_PER_SEC has *not* already been defined. But the file components/lora/node/lmic/oslmic.h includes sys/driver.h which includes luartos.h where `#define OSTICKS_PER_SEC 50000` is defined.

Hence, the line `#define OSTICKS_PER_SEC 32768` will never be used.
Please close this issue if your question has been answered.

gmsotavio, 2019-04-10:
@the0ne Thanks for answer my question.

Regards,
Otavio

closed 2019-04-10 by gmsotavio