whitecatboard/Lua-RTOS-ESP32 #185: Setting UART to 9 bits

andyrbarrow, 2018-07-28:
I see that you provide a means to set the UART to 9 bit data. Has this been tested? 

It appears to contradict the IDE documentation which does not provide a means of setting the UART for 9 bits.

the0ne, 2019-07-26:
Hi @andyrbarrow 
Lua-RTOS is meant as a multi-platform OS, hence the option to set different UART options.

ESP32 doesn't support 9 data bits. That's why the corresponding source code shows:
```
    switch (databits) {
        case 5: esp_databits = FIVE_BITS ; break;
        case 6: esp_databits = SIX_BITS  ; break;
        case 7: esp_databits = SEVEN_BITS; break;
        case 8: esp_databits = EIGHT_BITS; break;
        default:
            return driver_error(UART_DRIVER, UART_ERR_INVALID_DATA_BITS, NULL);
    }
```

This means that 9 data bits is not supported on the ESP32 platform and Lua-RTOS correctly throws an exception "INVALID_DATA_BITS".

the0ne, 2019-07-31:
@andyrbarrow feel free to re-open in case of questions

closed 2019-07-31 by the0ne