whitecatboard/Lua-RTOS-ESP32 #187: pack.unpack not working correct

thilo-hub, 2018-08-05:
Hi there

there are two issues I see with the pack packet:
The unpack has issues with small ascii characters ( unpack does not work - see example below)
// > s=string.char(5)
/ > print(pack.pack(s))
01400500
/ > print(pack.pack(pack.unpack("01400500")))
014000     
============ BAD RESULT ==========

The pack has issues with small ascii
/ > r=string.char(65,0,66)
/ > print( #r)
3
/ > print(pack.pack(r))
01404100
===== BAD RESULT =======

the0ne, 2019-03-11:
@thilo-hub I can reproduce both
first one looks like some kind of trim() issue
second one looks like truncation issue

jolivepetrus, 2019-03-16:
@thilo-hub, @the0ne,

Solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/f117ed11fe950287548b26246336df6cf0f78b53.

closed 2019-03-16 by jolivepetrus