whitecatboard/Lua-RTOS-ESP32 #191: BME680 support?

skitsanos, 2018-08-31:
Not an issue, but rather a question.

Is there any BME-680 support implemented in lua-rtos? and if yes, how do i access it?

Thanks.

jolivepetrus, 2018-10-07:
@skitsanos,

At this time, no, but it would be great!.

the0ne, 2019-04-09:
Looks like there is an implementation available: https://github.com/BoschSensortec/BME680_driver

elsbiet, 2019-04-16:
i'm slightly confuse because

    bme = sensor.attach("BME280", i2c.I2C0, 0x77, 400)
    ..
    temperature = bme:read("temperature")
    humidity = bme:read("humidity")
    pressure = bme:read("pressure")

used to work fine.

--> https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Sensor-module

the0ne, 2019-07-26:
@skitsanos are you able to integrate the [mentioned implementation](https://github.com/BoschSensortec/BME680_driver) into Lua-RTOS? Can you create a PR?

the0ne, 2022-12-02:
If you have the HW available, feel free to create a PR.

closed 2022-12-02 by the0ne