whitecatboard/Lua-RTOS-ESP32 #193: OpenVPN easy rsa in ESP32

aanand23, 2018-09-04:
Hi,

I need to generate RSA private key and certificate signing request in ESP32 and send the csr file to OpenVPN certificate authority for signature. My OpenVPN certificate authority is in docker container running on my local machine.

I am not able to generate RSA private key and certificate signing request in ESP32. Has anyone come across such issue? If so then could you please help me how to progress. Also please let me know whether this is possible or not?

I am already able to generate the private key, certificate signing request and sign the csr in OpenVPN CA ( using ./build-key client) but now I need to generate the key and csr file in ESP32 and sign it separately by OpenVPN CA to generate the public certificate.

Any help will be much appreciated!!!

Thanks

the0ne, 2019-04-09:
@aanand23 did you try to [generate and sign the keys](https://openvpn.net/community-resources/rsa-key-management/) on your pc and then [transfer them](https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/How-to-transfer-files) to the esp32?

the0ne, 2019-07-08:
@aanand23 were you able to fix your issue?

closed 2019-07-18 by the0ne