whitecatboard/Lua-RTOS-ESP32 #197: missing dependencies

Bazmundi, 2018-10-03:
Opting to build my own I installed tool chain per [instructions](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html).

I got no errors and everything seems to be in place.

However, I don't get directory paths for gettext or ncurses that even look remotely like the ones in env as below.

```
export LIBRARY_PATH="/usr/local/opt/gettext/lib:$LIBRARY_PATH";
export PKG_CONFIG_PATH="/usr/local/opt/ncurses/lib/pkgconfig:$PKG_CONFIG_PATH"
export CPATH="/usr/local/opt/gettext/include:$CPATH"

```

```
whereis ncurses returns : /usr/include/ncurses.h
whereis pkgconfig returns: /usr/share/pkgconfig
```

Now pkgconfig is also a python thingy that you can install separately which confused me somewhat.  I installed pkgconfig with pip but I am not sure if that added the /usr/share/pkgconfig folder and its gaff.

So, the upshot is that there is no /usr/local/opt folder on my machine post the install of tool chain.

I am running debian 9 stretch 64 bit AMD.

Any ideas please?

Cheers,
A

Bazmundi, 2018-10-03:
Hang on!  DOH!  Might have missed a step.  Will double check.

closed 2019-01-29 by jolivepetrus

Bazmundi, 2019-04-07:
Close with no help.  You must work for Microsoft!

the0ne, 2019-04-09:
@Bazmundi I think this was probably closed by accident. Did you manage to fix your issues? Please note that this project is developed by volunteers, not a company. So don't expect other people to solve your problems, although many will want to help you solve yours.

Make sure you've properly read the [howto](https://github.com/whitecatboard/Lua-RTOS-ESP32#method-2-build-by-yourself).
Make sure you've properly installed all prerequisites to build the [esp-idf](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html).
The above issue doesn't seem related to Lua-RTOS, so after going through the links provided you could also ask about your problem on stackoverflow or so.