whitecatboard/Lua-RTOS-ESP32 #198: lua editor keys clash when using screen via bash

Bazmundi, 2018-10-05:
Running screen as a terminal console via bash on debian stretch 9 amd64.

I can connect to a device using screen from a bash shell.
I can happily play with lua from repl.
I can get into editor on device but the ctrl-key combos used by the lua editor clash with the terminal cntrl-keys used by screen under bash.

Is there a known console based terminal people are using that allows all lua editor ctrl keys please?  In there a key file someone has sorted for screen perhaps?

Hold off suggesting miniterm via platformio as I have reported that Lua-RTOS-ESP32 crashes either miniterm or the platformio shell (or both).

Cheers,
A

jolivepetrus, 2018-10-05:
Hi @Bazmundi 

I always use picocom.

Can you attach the instructions to reproduce the crashes?.

Bazmundi, 2018-10-05:
@jolivepetrus 

https://community.platformio.org/t/serial-monitor-cracks-up-with-error/5259

I have tried running miniterm.py from command line but I am still working out the way to pump the baudrate into it.

closed 2019-02-13 by jolivepetrus