whitecatboard/Lua-RTOS-ESP32 #199: Enhancement - CSPlike channels

Bazmundi, 2018-10-05:
Since Lua-RTOS has tasks it would make sense to look at CSP like channels.

Starting postion might be loyso/LuaCSP perhaps.

jolivepetrus, 2018-10-07:
@Bazmundi,

Lua RTOS is a multi-thread Lua environment, which means that Lua scripts can run on multiple threads sharing the global state, so a global variable can be shared by multiple threads, and special Lua libraries are provided to protect concurrent access and synchronise threads:

* [Mutual exclusion functions](https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Thread-Module#mutual-exclusion-functions)
* [Events](https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/EVENT-module)

This is slightly different from other Lua concurrent approaches based on processes, in which each process runs a single-threaded Lua instance, each one with its own Lua state. In this approach, a communication mechanism between Lua instances is required to interchange data, because processes are isolated.

As far I can see, I'm not sure about the benefits of introducing a channel communication mechanism in Lua RTOS to share data between threads, because introduces a unnecessary overhead.

Bazmundi, 2018-10-08:
go lang site suggests:

In general use channels when Goroutines need to communicate with each other
and mutexes when only one Goroutine should access the critical section of
code.

So, it depends on use cases.
Events don't fit this use case in the absence of a payload.  A payload on
events might be of limited use.
If you are suggesting global variables with an event and mutexes to act as
a "channel" mechanism then you have your argument for channels.

Bazmundi, 2018-10-08:
A more detailed discussion is at:
http://www.flounder.com/no_synchronization.htm

Bazmundi, 2019-07-05:
If it kelps, even Kotlin has gone for channels.

rosarioculmone, 2020-11-04:
Events can be used to synchronize two tasks. For example between producer and consumer for the management of a buffer but I cannot use them. Have any of you tried them?

Bazmundi, 2020-11-05:
Events are not channels.