whitecatboard/Lua-RTOS-ESP32 #200: i2c has 44 byte memory leak

lu1210, 2018-10-06:
tprint(6,'              B-i2c start',os.stats('mem'))
hi2c:start()
hi2c:address(0x62,true)
status   = hi2c:read()
msb      = hi2c:read()
lsb      = hi2c:read()
hi2c:stop()
tprint(6,'              A-i2c stop ',os.stats('mem'))

    B-i2c start 71560
    A-i2c stop  71516

    B-i2c start 71528
    A-i2c stop  71484

jolivepetrus, 2018-10-07:
@lu1210,

I have published some things related to I2C, but nothing related to your issue.

I think that your issue is due to the Lua garbage collector. I usually detect memory leaks executing the suspect code in an infinite loop, that fails if a memory leak is present (not enough memory).

The following code is running hours and hours, and things seem right:

```
BMEx_addr = 0x76    -- I2C address
BMEx_reg_id = 0xd0  -- ID register

-- Setup I2C
BMEx = i2c.attach(i2c.I2C1, i2c.MASTER, 1000)
BMEx:setspeed(400)

-- Read BMEx ID register
while true do
  BMEx:start()
  BMEx:address(BMEx_addr, false)
  BMEx:write(BMEx_reg_id)
  BMEx:stop()
  BMEx:start()
  BMEx:address(BMEx_addr, true)
  id = BMEx:read()
  BMEx:stop()
end
```

lu1210, 2018-10-10:
You're right. That was a measurement error of mine. There was  a second thread in addition.

closed 2018-10-10 by lu1210