whitecatboard/Lua-RTOS-ESP32 #203: esp8266 support?

arrowcircle, 2018-10-25:
Hi!
Readme says Lua RTOS ESP32 support esp8266, but there are no boards available while flashing with wcc tool.
Is esp8266 supported?

solarkraft, 2019-02-06:
Any updates on this? Would it theoretically be compatible? Has there been a port?

hiperiondev, 2019-02-22:
All patches are for esp32 IDF

bhgv, 2019-03-18:
esp8266 has too (10x) low memory. there are was early choise for esp8266 but it doesn't work for me. 
i did a fork and remade it to run on low memory and some other features like work with ports extenders as esp8266 has too low pin/ports, small httpd, oled etc. but it's hard to work on so low memory.
if you want a workable kind of lua-rtos-esp8266 you may look to my github repo.

jolivepetrus, 2019-03-18:
@arrowcircle, @technomancy, @solarkraft, @hiperiondev, @bhgv,

Dear all,

In the past we migrate some parts of Lua RTOS running in PIC32MZ to ESP8266. This work was used later to create Lua RTOS for ESP32.

Due to the interest in Lua RTOS for ESP8266 we have decided to make repository public again. But there is a lot of work to do, and we need collaborators for it.

You can find the repository here:

https://github.com/whitecatboard/Lua-RTOS-ESP8266

bhgv, 2019-03-30:
> .. But there is a lot of work to do, and we need collaborators for it.
already working fork. a little remaden. include working edit, web-server, threads (but practically you can run only 1 thread additionally to main. low memory) etc

https://github.com/bhgv/LuOS9p-ESP8266

because esp8266 has too low memory you can't write big progs in Lua. the best - to write your own Lua modules in C and connect/configure them with Lua.

closed 2019-07-24 by jolivepetrus