whitecatboard/Lua-RTOS-ESP32 #204: Core1 panic'ed StoreProhibited

gjongenelen, 2018-10-25:
I'm getting random panics when reading data via UART.

[image]

How can I debug this?

jolivepetrus, 2018-10-30:
@gjongenelen,

Please, proceed as follows:

make clean
make upgrade-idf
make

gjongenelen, 2018-10-31:
Did that several times, but without any success.
I found out that it is the *el/*l-argument which is causing the issue.

```
uart.read(uart.UART1, "*el", 1000); -- StoreProhibited
uart.read(uart.UART1, "*l", 1000); -- StoreProhibited
uart.read(uart.UART1, "*c", 1000); -- No problems (so far)
```
https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/449517900a2ee4167628fc4f8547ce240e9de23d/components/lua/modules/hw/uart.c#L275

the0ne, 2018-11-07:
@gjongenelen you should make sure that `str` is not null after malloc, before using it in line 278.

https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/449517900a2ee4167628fc4f8547ce240e9de23d/components/lua/modules/hw/uart.c#L275-L278

the0ne, 2019-02-10:
fixed in 9e61e52725c4b77574e9bf9261a96f918ab779f5
as it currently only is in [my fork](https://github.com/the0ne/Lua-RTOS-ESP32) and the PR #184 you may want to cherry-pick it.

jolivepetrus, 2019-02-11:
@gjongenelen, @the0ne,

Merged in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/2b98d995192cc448af0499957635c5abbe8e4dea.

closed 2019-02-11 by jolivepetrus