whitecatboard/Lua-RTOS-ESP32 #206: panic reboot happens

lu1210, 2018-10-28:
prassertion "pbuf_free: p->ref > 0" failed: file "/home/fh/esp/esp-idf/components/lwip/lwip/src/core/pbuf.c", line 765, function: pbuf_free
abort() was called at PC 0x400d3e83 on core 1

backtrace: 0x40090533:0x3ffe7260 0x400907f9:0x3ffe7280 0x400d3e83:0x3ffe72a0 0x4014897f:0x3ffe72d0 0x401442a3:0x3ffe72f0 0x40144de6:0x3ffe7370 0x40144e42:0x3ffe73a0 0x400f9ad1:0x3ffe73c0 0x400fa94e:0x3ffe73f0 0x400fabb1:0x3ffe7420 0x400f944d:0x3ffe7c70 0x400f47d9:0x3ffe7c90 0x400ed00d:0x3ffe7cc0 0x400f49b5:0x3ffe7d10 0x400f49d5:0x3ffe7d30 0x400f4c39:0x3ffe7d50 0x400f4491:0x3ffe7d70 0x400f4a02:0x3ffe7df0 0x400f5aa5:0x3ffe7e20 0x400ed620:0x3ffe7e50 0x400f47d9:0x3ffe7e70 0x400ed00d:0x3ffe7ea0 0x400f49b5:0x3ffe7ef0 0x400f49d5:0x3ffe7f10 0x400f5a4c:0x3ffe7f30 0x400edf45:0x3ffe7f50 0x400f47d9:0x3ffe7f70 0x400f49ae:0x3ffe7fa0 0x400f49d5:0x3ffe7fc0 0x400f4c39:0x3ffe7fe0 0x400f4491:0x3ffe8000 0x400f4a02:0x3ffe8080 0x400f5aa5:0x3ffe80b0 0x400ed620:0x3ffe80e0 0x400f47d9:0x3ffe8100 0x400ed00d:0x3ffe8130 0x400f49b5:0x3ffe8180 0x400f49d5:0x3ffe81a0 0x400f4c39:0x3ffe81c0 0x400f4491:0x3ffe81e0 0x400f4a02:0x3ffe8260 0x400f5aa5:0x3ffe8290 0x400e3bd5:0x3ffe82c0 0x400d496f:0x3ffe82e0
0x40090533: invoke_abort at panic.c:678
0x400907f9: abort at panic.c:678
0x400d3e83: __assert_func at assert.c:63 (discriminator 8)
0x4014897f: pbuf_free at pbuf.c:1331
0x401442a3: lwip_recvfrom at sockets.c:3471
0x40144de6: lwip_recvfrom_r at sockets.c:3471
0x40144e42: lwip_recv_r at sockets.c:3471
0x400f9ad1: recv at usocket.c:357
 (inlined by) socket_recv at usocket.c:260
0x400fa94e: buffer_get at buffer.c:266
0x400fabb1: recvline at buffer.c:229
 (inlined by) buffer_meth_receive at buffer.c:121
0x400f944d: meth_receive at tcp.c:135
0x400f47d9: luaD_precall at ldo.c:729
0x400ed00d: luaV_execute at lvm.c:1260
0x400f49b5: luaD_call at ldo.c:729
0x400f49d5: luaD_callnoyield at ldo.c:729
0x400f4c39: f_call at lapi.c:1222
0x400f4491: luaD_rawrunprotected at ldo.c:729
0x400f4a02: luaD_pcall at ldo.c:740
0x400f5aa5: lua_pcallk at lapi.c:1222
0x400ed620: luaB_pcall at lbaselib_adds.inc:88
0x400f47d9: luaD_precall at ldo.c:729
0x400ed00d: luaV_execute at lvm.c:1260
0x400f49b5: luaD_call at ldo.c:729
0x400f49d5: luaD_callnoyield at ldo.c:729
0x400f5a4c: lua_callk at lapi.c:1222
0x400edf45: luaB_dofile at lbaselib_adds.inc:88
0x400f47d9: luaD_precall at ldo.c:729
0x400f49ae: luaD_call at ldo.c:729
0x400f49d5: luaD_callnoyield at ldo.c:729
0x400f4c39: f_call at lapi.c:1222
0x400f4491: luaD_rawrunprotected at ldo.c:729
0x400f4a02: luaD_pcall at ldo.c:740
0x400f5aa5: lua_pcallk at lapi.c:1222
0x400ed620: luaB_pcall at lbaselib_adds.inc:88
0x400f47d9: luaD_precall at ldo.c:729
0x400ed00d: luaV_execute at lvm.c:1260
0x400f49b5: luaD_call at ldo.c:729
0x400f49d5: luaD_callnoyield at ldo.c:729
0x400f4c39: f_call at lapi.c:1222
0x400f4491: luaD_rawrunprotected at ldo.c:729
0x400f4a02: luaD_pcall at ldo.c:740
0x400f5aa5: lua_pcallk at lapi.c:1222
0x400e3bd5: lthread_start_task at thread.c:147
0x400d496f: pthreadTask at _pthread.c:551

lu1210, 2018-10-29:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400fe122  PS      : 0x00060230  A0      : 0x8015bddc  A1      : 0x3ffe70f0  
A2      : 0x00000000  A3      : 0x3ffe7124  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffe7370  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000003  
A10     : 0x00000000  A11     : 0x3ffe713c  A12     : 0x3ffe7144  A13     : 0x3ffe714c  
A14     : 0x00000001  A15     : 0xffffffff  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000006b  LBEG    : 0x400fd6b5  LEND    : 0x400fd6bf  LCOUNT  : 0x00000000  

Backtrace: 0x400fe122:0x3ffe70f0 0x4015bdd9:0x3ffe7350 0x400f980c:0x3ffe7370 0x400f9af6:0x3ffe73c0 0x400fa94e:0x3ffe73f0 0x400fabb1:0x3ffe7420 0x400f944d:0x3ffe7c70 0x400f47d9:0x3ffe7c90 0x400ed00d:0x3ffe7cc0 0x400f49b5:0x3ffe7d10 0x400f49d5:0x3ffe7d30 0x400f4c39:0x3ffe7d50 0x400f4491:0x3ffe7d70 0x400f4a02:0x3ffe7df0 0x400f5aa5:0x3ffe7e20 0x400ed620:0x3ffe7e50 0x400f47d9:0x3ffe7e70 0x400ed00d:0x3ffe7ea0 0x400f49b5:0x3ffe7ef0 0x400f49d5:0x3ffe7f10 0x400f5a4c:0x3ffe7f30 0x400edf45:0x3ffe7f50 0x400f47d9:0x3ffe7f70 0x400f49ae:0x3ffe7fa0 0x400f49d5:0x3ffe7fc0 0x400f4c39:0x3ffe7fe0 0x400f4491:0x3ffe8000 0x400f4a02:0x3ffe8080 0x400f5aa5:0x3ffe80b0 0x400ed620:0x3ffe80e0 0x400f47d9:0x3ffe8100 0x400ed00d:0x3ffe8130 0x400f49b5:0x3ffe8180 0x400f49d5:0x3ffe81a0 0x400f4c39:0x3ffe81c0 0x400f4491:0x3ffe81e0 0x400f4a02:0x3ffe8260 0x400f5aa5:0x3ffe8290 0x400e3bd5:0x3ffe82c0 0x400d496f:0x3ffe82e0
0x400fe122: esp_vfs_select at vfs.c:924
0x4015bdd9: select at select.c:62
0x400f980c: socket_waitfd at usocket.c:357
0x400f9af6: socket_recv at usocket.c:357
0x400fa94e: buffer_get at buffer.c:266
0x400fabb1: recvline at buffer.c:229
 (inlined by) buffer_meth_receive at buffer.c:121
0x400f944d: meth_receive at tcp.c:135
0x400f47d9: luaD_precall at ldo.c:729
0x400ed00d: luaV_execute at lvm.c:1260
0x400f49b5: luaD_call at ldo.c:729
0x400f49d5: luaD_callnoyield at ldo.c:729
0x400f4c39: f_call at lapi.c:1222
0x400f4491: luaD_rawrunprotected at ldo.c:729
0x400f4a02: luaD_pcall at ldo.c:740
0x400f5aa5: lua_pcallk at lapi.c:1222
0x400ed620: luaB_pcall at lbaselib_adds.inc:88
0x400f47d9: luaD_precall at ldo.c:729
0x400ed00d: luaV_execute at lvm.c:1260
0x400f49b5: luaD_call at ldo.c:729
0x400f49d5: luaD_callnoyield at ldo.c:729
0x400f5a4c: lua_callk at lapi.c:1222
0x400edf45: luaB_dofile at lbaselib_adds.inc:88
0x400f47d9: luaD_precall at ldo.c:729
0x400f49ae: luaD_call at ldo.c:729
0x400f49d5: luaD_callnoyield at ldo.c:729
0x400f4c39: f_call at lapi.c:1222
0x400f4491: luaD_rawrunprotected at ldo.c:729
0x400f4a02: luaD_pcall at ldo.c:740
0x400f5aa5: lua_pcallk at lapi.c:1222
0x400ed620: luaB_pcall at lbaselib_adds.inc:88
0x400f47d9: luaD_precall at ldo.c:729
0x400ed00d: luaV_execute at lvm.c:1260
0x400f49b5: luaD_call at ldo.c:729
0x400f49d5: luaD_callnoyield at ldo.c:729
0x400f4c39: f_call at lapi.c:1222
0x400f4491: luaD_rawrunprotected at ldo.c:729
0x400f4a02: luaD_pcall at ldo.c:740
0x400f5aa5: lua_pcallk at lapi.c:1222
0x400e3bd5: lthread_start_task at thread.c:147
0x400d496f: pthreadTask at _pthread.c:551

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:3856
ho 0 tail 12 room 4
load:0x40078000,len:7696
load:0x40080400,len:6240
entry 0x400806f0
Booting Lua RTOS...

  /\       /\

 /  \_____/  \

/_____________\

W H I T E C A T

jolivepetrus, 2018-12-04:
@lu1210,

Any feedback on this issue?.

lu1210, 2018-12-05:
Since no panic message has occurred for 10 days the issue is closed.

closed 2018-12-05 by lu1210