whitecatboard/Lua-RTOS-ESP32 #209: Httpsrv.c LoadProhibited

gjongenelen, 2018-11-05:
We're getting LoadProhibited errors in the http-server. It looks like the server is unable to close the client in https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/e414c76583a8208d3d5cab1acc645910f15f04ef/components/http/httpsrv.c#L1162.

Uncommenting lines 1162, 1165 and 1166 prevent the error occurring but make the http-server unusable when 10 requests has taken place. (socket error)
[image]

jolivepetrus, 2018-12-03:
@gjongenelen,

Any feedback on this issue?.

gjongenelen, 2018-12-06:
Couldn’t test it yet. Had to switch to c++ because I had to deliver the prototype. 
Will test it when I have the time