whitecatboard/Lua-RTOS-ESP32 #210: MQTT cannot connect

ctshh, 2018-11-18:
The preamble:
```
/ > os.version()
Lua RTOS        beta 0.1        1528456632      80412d18bd82a8f01056e2e688a0aef30cdc2bec
/ > os.cpu()
ESP32 rev 1 rev A1
/ > netstat
wf: mac address 24:0a:c4:xx:xx:xx
   ip address 10.3.2.170 / netmask 255.255.255.0
   gw address 10.3.2.1
  ip6 address fe80:0000:0000:0000:260a:c4xx:xxxx:xxxx

en: mac address 00:00:00:00:00:00
   ip address 0.0.0.0 netmask 0.0.0.0
   gw address 0.0.0.0

/ > net.ping('test.mosquitto.org')
PING test.mosquitto.org (37.187.106.16): 32 data bytes
60 bytes from 37.187.106.16: icmp_seq=1 time=24.573 ms
60 bytes from 37.187.106.16: icmp_seq=2 time=27.957 ms
60 bytes from 37.187.106.16: icmp_seq=3 time=27.790 ms
60 bytes from 37.187.106.16: icmp_seq=4 time=30.657 ms
4 packets transmitted, 4 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 24.573/27.744/30.657/2.156 ms
/ >
``` 

Now, setting up an MQTT connection works, opening the connection leads to a timeout:
```
/ > client = mqtt.client('somerandomgasfd', 'test.mosquitto.org', 1883, false)
/ > client:connect("","")
stdin:1: 201326594:can't connect (timeout)
stack traceback:
        [C]: in method 'connect'
        stdin:1: in main chunk
        [C]: in ?
/ >
```
Same goes for a secure MQTT connection:
```
/ > client = mqtt.client('myespkjasda', 'test.mosquitto.org', 8883, true)
/ > client:connect("","")
stdin:1: 201326594:can't connect (timeout)
stack traceback:
        [C]: in method 'connect'
        stdin:1: in main chunk
        [C]: in ?
/ >

```

Any idea why my MQTT connection is failing so miserably? The same connection from my notebook works fine.

Also and on a sidenote, when using the WebIDE and configuring the MQTT session, "Get data" does get the server info, yet no suername or password shows up. Any ideas on this?

Thank you for any help,
 Christian

jolivepetrus, 2018-11-19:
@ctshh,

Thanks for report this. Solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/e6dd5c7d218fcd30569addc54fce7fb3a78ee4b5.

ctshh, 2018-11-28:
I've flashed the current version (80412d18bd82a8f01056e2e688a0aef30cdc2bec), uplaoded the attached sketch, yet I still get a timeout arror.
Do I have to build the current firmware myself or shoud this be automatic?

ctshh, 2018-11-28:
Err. no close! Sorry...

ctshh, 2018-11-28:
Connection to IBM Watson IoT Quickstart fails, too.
The "ID" MQTT parameter is not supported (which ahs nothing to do with the connection failing, it's just something we might want to take into consideration that is exists. If teh ID field would be supported, a very easy Quickstart with the IBM platform could be sketched as a nice example.

jolivepetrus, 2018-11-29:
@ctshh,

Yes, the best way to ensure to test last changes is to build Lua RTOS by yourself. It may be a delay between publish an update to github, and the official Lua RTOS builds that can be flashed with the wcc tool.

About IBM Watson, can you provide your sample code, with the credentials you use? . You can provide temporary credentials, or send me the credentials by private message to jolive@whitecatboard.org. This should be great, and faster than create a IBM Watson account.

ctshh, 2018-11-30:
`https://quickstart.internetofthings.ibmcloud.com/` can be used without credentials:

`https://quickstart.internetofthings.ibmcloud.com/#/device/whitecat/sensor/` shows what `mosquitto_pub -h quickstart.messaging.internetofthings.ibmcloud.com -p 1883 -t "iot-2/evt/helloworld/fmt/json" -i "d:quickstart:mosquitto:whitecat" -m '{"helloworld": 12}'` on the console writes; `mosquitto_sub -h quickstart.messaging.internetofthings.ibmcloud.com -p 1883 -i "a:quickstart:3141592" -t "iot-2/type/mosquitto/id/whitecat/evt/helloworld/fmt/json"` shows the output on the commandline.

jolivepetrus, 2018-12-03:
@ctshh,

Solved in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/2703dd1ec8dfcb6ae9bcb38d4b48fddd3bdbfa2b.