whitecatboard/Lua-RTOS-ESP32 #215: Firmware crashing on lua_pushnmber

xopxe, 2018-12-13:
I'm passing a lua_Number from C to a Lua callback with something like

              lua_pushnumber(TL, (lua_Number) 2.0/3);

The Lua callback is called succesfuly, but the system crashes shortly after with

` ` `
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4009503e  PS      : 0x00060830  A0      : 0x00000000  A1      : 0x3ffbc910  
A2      : 0x00060c30  A3      : 0x00001e6a  A4      : 0x00000009  A5      : 0x3ffc1984  
A6      : 0x3ffc1980  A7      : 0x00001e66  A8      : 0x80095100  A9      : 0x3ffbc8d0  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffbc0f8  A13     : 0x00000000  
A14     : 0x3ffbc120  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00060c44  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4009503e:0x3ffbc910 0x7ffffffd:0x3ffbc940

Rebooting...
```
I'm passing integers and strings with no problem, only see this with push_number.

the0ne, 2019-04-09:
@xopxe any progress / solution on this? does it still occur? how can the issue be reproduced?

xopxe, 2019-04-12:
I wont be able to check if I can reproduce this for a couple weeks, but I will. If you prefer you can close it and I'll reopen if needed.

the0ne, 2019-07-08:
@xopxe please re-open once you re-checked

closed 2019-07-08 by the0ne