whitecatboard/Lua-RTOS-ESP32 #216: SPI buffering code fails when mixing write and readwrite in same cs

tcoram, 2019-01-10:
I am working with a delta-sigma ADC chip (TI ads1120) that talks SPI.  I am sending a mix of write() and readwrite() and my chip goes into weird modes.

Here is what my code looks like:

```
function ads.rdata()
   ads.dev:select()
   ads.dev:write(16)
   local val = ads.dev:readwrite(0,0)
   ads.dev:deselect()
   return val[1] << 8 | val[2]
end
```

Here is my SPI configuration:

`ads.dev = spi.attach(ads.spi, spi.MASTER, ads.ns, 1000000, 8, 1)`

Using my Saleae Logic Analyzer, I can see that 4 (instead of 3) master transactions are being sent.  The final (extra/spurious) one sends the last byte read (which unfortunately sends my ads chip into a weird state).  Here is what my analyzer sees (decoded):

> MOSI: '16' (0x10);  MISO: 0 (0x30)
> MOSI: '0' (0x00);  MISO: 0 (0x30)
> MOSI: '0' (0x10);  MISO: '18' (0x12)
> MOSI: '18' (0x12);  MISO: '255' (0xFF)

If I modify lua/modules/hw/spi.c:lspi_rw_helper()  to remove the buffering (and write individual bytes) then the issue goes away and the same above Lua code gives:

> MOSI: '16' (0x10);  MISO: 0 (0x30)
> MOSI: '0' (0x00);  MISO: 0 (0x30)
> MOSI: '0' (0x10);  MISO: '18' (0x12)

the0ne, 2019-07-08:
@tcoram were you able to locate the issue? From your description it looks like there could be some issue with the buffer handling in spi.c, right?

Mynogs, 2020-04-22:
Please look at these pull requests:
Fix for issue #345 and #337

the0ne, 2022-12-02:
@Mynogs thanks for your PRs!

closed 2022-12-02 by the0ne