whitecatboard/Lua-RTOS-ESP32 #221: ublox NINA W

tcpipchip, 2019-01-28:
Sir, 

I see that LUA RTOS image is too big to fit a ESP32 with 2MB flash!

Can you give some suggestion on "make menuconfig" to reduce the size of LUA RTOS  to feet on ESP32 (2MB version) ?

No problem if  i loose some resource!

jolivepetrus, 2019-01-29:
@tcpipchip,

Maybe you can start disabling graphic displays, and SSH server in Kconfig:

Component config > Lua RTOS > Lua > Lua RTOS Modules
Component config > Lua RTOS > Network services > SSH server

tcpipchip, 2019-01-29:
HI @jolivepetrus 
Thanks to feedback!
Still not solve it the size! I wil try FS size space reduce!

jolivepetrus, 2019-02-06:
@tcpipchip,

Any progress in this issue?

tcpipchip, 2019-02-06:
Hello Petrus
Yes, it´s running on NINA W.
But WHITECAT not works! Always shutdown when try to connect to LUA RTOS.
here the BLOG
https://lua-ninaw.blogspot.com/2019/01/nina-w-programado-em-lua.html
Thanks to remember me! And thanks for your help!

jolivepetrus, 2019-02-06:
@tcpipchip,

When you say "but WHITECAT not works", you mean that you can't program the board with The Whitecat IDE?

tcpipchip, 2019-02-06:
yes,
The Whitecat agent closes when ide try to contact it..
How the whitecat search for a the LUA RTOS ?
Today i am using the LUA RTOS with TERATERM....

tcpipchip, 2019-02-06:
I use WINDOWS 7

jolivepetrus, 2019-02-12:
@tcpipchip,

To add this board and see what happens I should need the following:

* Know the VID:PID of the board's usb to serial adapter. You can get this information executing the following in the command line:

python -m serial.tools.list_ports -v

* Have a copy of your sdkconfig file. You can send it to jolive@whitecatboard.org

* If you have the possibility it would be nice to have one of this board.

tcpipchip, 2019-02-12:
Good morning Jaume

/dev/ttyUSB3
    desc: EVB-NINA-W1
    hwid: USB VID:PID=0403:6011 SER=D4CA6EB4CDC1 LOCATION=1-1:1.3

Attached sdkconfig

Btw, do mean about module NINA W, right ?

  *   Have a copy of your sdkconfig file. You can send it to jolive@whitecatboard.org<mailto:jolive@whitecatboard.org>

jolivepetrus, 2019-02-13:
@tcpipchip,

Initial support for this boar has been added in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/70d4160f9ebc1de38ae716308204e28fc69d84f0.

It should work now with Whitecat IDE.

About the possibility to have one of this board, I mean a board like than the board you have. This help us to be more efficient and faster when resolving issues.

tcpipchip, 2019-02-13:
Hi Jaume

Thank you so much! i will test soon!

Where are you in EUROPE ?

I will talk with distributor and see how can we send you a sample, ok ?

Miguel

PS: btw, are you thinking to add the basic BLE ?

jolivepetrus, 2019-02-13:
@tcpipchip,

Yes, we are in Europe. Our contact address is:

whitecatboard.org
PLAÇA CAN SURTIS S/N (EDIFICI CITILAB CORNELLÀ)
08940 - CORNELLÀ DE LLOBREGAT
BARCELONA

About BLE it's not planned yet. No much funds now, so Whitecat is mainly financed with private funds.

Best regards.

jolivepetrus, 2019-02-13:
@tcpipchip,

Sorry, a transcription error ....

whitecatboard.org
PLAÇA CAN SURIS S/N (EDIFICI CITILAB CORNELLÀ)
08940 - CORNELLÀ DE LLOBREGAT
BARCELONA

tcpipchip, 2019-02-13:
Good morning!!!!

Still not works! Can find the NINA W102

Serial is working, because i can see the LUA BOOT.

Tested on minicom (linux)

I gave /dev/ttyUSBx permissions

tcpipchip, 2019-02-13:
I will verify!

tcpipchip, 2019-02-15:
I published now on UBLOX forum about LUA RTOS

tcpipchip, 2019-02-15:
can you modify again ?
/dev/ttyUSB2        
    desc: EVB-NINA-W1
    hwid: USB VID:PID=0403:6011 SER=D4CA6EB4CDC1 LOCATION=1-1:1.2

tcpipchip, 2019-02-15:
Hi Petrus
I guess that i found a bug
After execute make, the script ask me the EVK-NINA-W ID, i choose 31...and all compile...
When i burn and execute the LUA, not shows EVK-NINA-W on boot, but generic.
And in the LINUX cant find BOARD, try SCANNING, but no boards.

tcpipchip, 2019-02-15:
looks that is looking for PID generic board but mine is EVK-NINA-W

jolivepetrus, 2019-02-28:
@tcpipchip,

VID/PID for your board has been added to The Whitecat IDE. It should work now.

tcpipchip, 2019-02-28:
Thank you! Btw i dont forget about your sample!

tcpipchip, 2019-02-28:
Still not work! Testing on linux vm! Do you try to send some uart package to lua board to answer ?
I will try in other machine

Enviado do meu iPhone

tcpipchip, 2019-03-01:
Hi, i tested in other machine, W7,  same problem of other W7, agent execute, show scanning and disapear!

In LINUX VM only scanning and shows a strange string sequence and again the SCANNING message

Strange!

Did you try in W7 or only 10 ?

jolivepetrus, 2019-03-02:
@tcpipchip,

I have an idea about what happens.

The following is a capture of my board's console output when doing a manual reset:

`
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:4016
ho 0 tail 12 room 4
load:0x40078000,len:9288
ho 0 tail 12 room 4
load:0x40080400,len:5928
entry 0x400806bc
Booting Lua RTOS...
`

Can you atached a capture console of your board when doing a manual reset. Whitecat Create Agent uses the bootloader output to take some decisions about the board. Maybe your board has a pull-down on GPIO15 that disables the bootloader output. If this is your case, we will write some changes in the Whitecat Create Agent.

tcpipchip, 2019-03-02:
Hi!!!!

I will send you this info Tuesday

Btw, dont whitecat send some txd data first to serial ? Or first wait for a rxd data ?

jolivepetrus, 2019-03-02:
@tcpipchip,

First data is sent to the board only if Lua RTOS is detected on it.

tcpipchip, 2019-03-02:
Hummm
Make sense!

Enviado do meu iPhone

tcpipchip, 2019-03-02:
Make sense!!!!

tcpipchip, 2019-03-05:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4148
load:0x40078000,len:9748
ho 0 tail 12 room 4
load:0x40080400,len:6060
entry 0x400806dc
Booting Lua RTOS...

[2J[1;1H  /\       /\

 /  \_____/  \

/_____________\

W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1550248939

commit 9c757a84beaa5e5cdd6a3605cb09de5d90bb59c1

Running from factory partition

board type GENERIC

cpu ESP32 rev 1 at 240 Mhz

spiffs start address at 0xf000, size 256 Kb

spiffs mounted on /

Lua RTOS beta 0.1 powered by Lua 5.3.4

/ >

tcpipchip, 2019-03-17:
hi, no news there ?

jolivepetrus, 2019-03-18:
@tcpipchip,

Well I have found some issues related with sensor API is not enabled for this board. I have just update the prerequisites in the IDE to solve id.

Another important thing is that you should see something like the following in your console:

```
build 1552906962
commit 7eb7045fc7bf91dc763ef5bcfe5f1ea2648f75cb
Running from factory partition
board type EVK-NINA-W
cpu ESP32 rev 1 at 240 Mhz, 2 Cores, flash memory, 2.4GHz WiFi, Bluetooth LE, Bluetooth Classic
```

board type EVK-NINA-W, if you can't see this message in your console you should rebuild Lua RTOS from the scratch.

I haven't this board, but flashing EVK-NINA-W firmware in one of our boards, we can see the board in the IDE:

[image]

tcpipchip, 2019-03-18:
Hi Jaume

Show me

 /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1552406860
commit 9c757a84beaa5e5cdd6a3605cb09de5d90bb59c1
Running from factory partition
board type GENERIC
cpu ESP32 rev 1 at 240 Mhz
spiffs start address at 0xf000, size 256 Kb
spiffs mounted on /

Lua RTOS beta 0.1 powered by Lua 5.3.4

/ >

Can you instruct me how to update the git and compile ?

[image]<https://user-images.githubusercontent.com/16651312/54526930-9d91ec00-4978-11e9-8613-cc393fbe1a35.png>

[https://ipmcdn.avast.com/images/icons/icon-envelope-tick-round-orange-animated-no-repeat-v1.gif]<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>  Livre de vírus. www.avast.com<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.

tcpipchip, 2019-03-18:
got this :)

 /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1552912499
commit 7eb7045fc7bf91dc763ef5bcfe5f1ea2648f75cb
Running from factory partition
board type EVK-NINA-W
cpu ESP32 rev 1 at 240 Mhz, 2 Cores, 2.4GHz WiFi, Bluetooth LE, Bluetooth Classic
spiffs start address at 0xf000, size 256 Kb
spiffs mounted on /

Lua RTOS beta 0.1 powered by Lua 5.3.4

I will test now the WHITECAT :)

tcpipchip, 2019-03-18:
If not, something in the linux is blocking the rxd of my board!

minicom for example, can´t transmit

jolivepetrus, 2019-03-18:
sudo usermod -a -G dialout your_user?

tcpipchip, 2019-03-18:
but no blocking to putty

tcpipchip, 2019-03-18:
show this

tcpipchip, 2019-03-18:
Show thishttps://furb-my.sharepoint.com/:i:/g/personal/maw_furb_br/ERT3EtBS7tpGlflg_qXLzzQBzwShF0Ejupo-jZ72UFtX6Q?e=8b8WRH

jolivepetrus, 2019-03-18:
@tcpipchip,

Restart all, and try again.
Do you have some tty session opened while the IDE is opened?

tcpipchip, 2019-03-18:
I did!

Not open tty session at the same time!

Late i will test in a windows 10 and not VM linux

tcpipchip, 2019-03-18:
which OS are you using ?

tcpipchip, 2019-03-19:
Jaume
I tested in other windows machine and i see something interesting...
After reload the whitecat agent i see the CTS toggle, then i imediately press the reset and showed NINA W on whitecat ide! I got this after repeat this process many times!
Looks that is a syncronize problem!

Enviado do meu iPhone

tcpipchip, 2019-03-19:
Tomorrow i will connect cts to reset to make new tests!

Enviado do meu iPhone

tcpipchip, 2019-03-19:
Jaume
can you do a special  AGENT version that toggle the DSR before try to communicate with NINA W102 ?
Looking the
https://www.u-blox.com/sites/default/files/EVK-NINA-W1-EVK-NINA-B2_UserGuide_%28UBX-17011007%29.pdf
page 16, DSR is resetting the MODULE! (up right side)

tcpipchip, 2019-03-20:
If possible, ok ?

jolivepetrus, 2019-03-26:
@tcpipchip,

It will be possible, I hope, but time it's a finite resource.

To clarify things, what is exactly the problem with DSR?

Now, the create agent toggle the RTS signal (OFF, ON, OFF) to reset the board:

```
	// Reset board
	options := serial.RawOptions
	options.BitRate = 115200
	options.Mode = serial.MODE_READ_WRITE

	options.RTS = serial.RTS_OFF
	board.port.Apply(&options)

	time.Sleep(time.Millisecond * 10)

	options.RTS = serial.RTS_ON
	board.port.Apply(&options)

	time.Sleep(time.Millisecond * 10)

	options.RTS = serial.RTS_OFF
	board.port.Apply(&options)
```

To propose a modification on the create agent to met your requirements it would be a help to get clear and exact requirements.

tcpipchip, 2019-03-26:
Hi Jaume

I asked you about DSR because in the EVK-NINA-W kit they use the DSR to reset the board

[cid:4e0576d5-35fa-48be-9ba7-85c123e070b2]

You can see that! Update when you have time and i test!

Thanks!

Miguel Wisintainer

tcpipchip, 2019-08-20:
Petrus,
Are you there ?

Miguel

[https://ipmcdn.avast.com/images/icons/icon-envelope-tick-round-orange-animated-no-repeat-v1.gif]<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>  Livre de vírus. www.avast.com<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.

tcpipchip, 2020-10-08:
Hi, 
I am backing to make the EVK-NINA-W102 works with WHITECAT
https://lua-ninaw.blogspot.com/2019/01/nina-w-programado-em-lua.html
Making changes on BLOG

tcpipchip, 2020-10-08:
@jolivepetrus 

Can you modify the Whitecat agent to use the DSR to make the reset board and not RTS ?
[image]
Or instruct me ....
Thanks!

tcpipchip, 2020-10-12:
i got some progress! I soldered the boar to make the RTS the RESET, but the WHITECAT IDE always now is stopping on DOWNLOADING FRAMEWORK message

the0ne, 2022-12-02:
Feel free to reopen in case of questions.

tcpipchip, 2022-12-02:
Wich are the news ?

tcpipchip, 2022-12-02:
are you the teacher ?

tcpipchip, 2022-12-02:
still asking to VID and PID ?