whitecatboard/Lua-RTOS-ESP32 #222: How can I attach an external gpio extender

lu1210, 2019-01-29:
Hi, I have to extend my system with an IO expander e.g. PCA8574 for i2c bus. Can I attach the 8 additional gpio's to the system and in which way :-)?

jolivepetrus, 2019-01-29:
@lu1210,

Lua RTOS has support for the PCA9698 (4O I/O NO PULLUP) and PCA9505 series (40 I/O PULLUP). Both chips needs I2C (SDA / SCL), plus any GPIO to get the interrupts.

When enabled in Lua RTOS firmware, the use is transparent to the user, because the access to this feature is done trough the Lua pio module (see https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/PIO-Module). External GPIO numbers start at 40, so 40 is external GPIO 0, 41 is external GPIO 1, and so on ....

Maybe you can adapt https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/components/sys/drivers/pca9xxx.c to met your needs.

lu1210, 2019-01-30:
thanks for the fast answer. I have found an other way to solve my requirement with the function pio.pin.interrupt().

closed 2019-01-30 by lu1210