whitecatboard/Lua-RTOS-ESP32 #224: ESP32 crashes after 84-th UART read

xPaRi, 2019-02-09:
Hi, 
after 84 uart.read(...) from non active port is ESP32 restarting. 

In real operation, there are also valid read operations, but when the total number of 84 is reached, a restart occurs.

TRY-CATCH did not help.

Example:

index = 0
uart.attach(uart.UART1, 9600, 8, uart.PARNONE, uart.STOP1)

print("START")

while(true) do
    s = uart.read(uart.UART1, "*el", 100)

    index = index + 1
    print(index)
end

the0ne, 2019-02-10:
confirmed

```
[...]
90
91
92
93
94
<MALLOC_DEBUG>	WARNING: Out of memory! Requested size 2048+48. Returning NULL (in /home/.../esp32/whitecatboard/Lua-RTOS-ESP32/components/lua/modules/hw/uart.c:291)
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e0cd3  PS      : 0x00060e30  A0      : 0x800e506b  A1      : 0x3ffc5770  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffb3b58  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800e0c97  A9      : 0x00000064  
A10     : 0x00000000  A11     : 0x3ffc578b  A12     : 0x00000064  A13     : 0x7fef069b  
A14     : 0x3f40aa48  A15     : 0x00000008  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400fb53e  LEND    : 0x400fb561  LCOUNT  : 0x00000000  

Backtrace: 0x400e0cd3:0x3ffc5770 0x400e5068:0x3ffc57b0 0x40101b91:0x3ffc57e0 0x400f64fd:0x3ffc5820 0x40101e36:0x3ffc5880 0x40101e79:0x3ffc58b0 0x4010388c:0x3ffc58e0 0x400f7cc4:0x3ffc5910 0x40101b91:0x3ffc5940 0x400f64fd:0x3ffc5980 0x40101e36:0x3ffc59e0 0x40101e79:0x3ffc5a10 0x40102375:0x3ffc5a40 0x401016b5:0x3ffc5a70 0x40102090:0x3ffc5af0 0x4010390a:0x3ffc5b30 0x400fac1b:0x3ffc5b70 0x400faeef:0x3ffc5ba0 0x400fb285:0x3ffc5bd0 0x40101b91:0x3ffc5c00 0x40101e2f:0x3ffc5c40 0x40101e79:0x3ffc5c70 0x40102375:0x3ffc5ca0 0x401016b5:0x3ffc5cd0 0x40102090:0x3ffc5d50 0x4010390a:0x3ffc5d90 0x400fafcf:0x3ffc5dd0 0x400d5e95:0x3ffc5e00 0x400d55a3:0x3ffc5e30

Rebooting...

```

the0ne, 2019-02-10:
fixed in 9e61e52725c4b77574e9bf9261a96f918ab779f5
as it currently only is in [my fork](https://github.com/the0ne/Lua-RTOS-ESP32) and the PR #184 you may want to cherry-pick it.

jolivepetrus, 2019-02-11:
@xPaRi, @the0ne,

Merged in https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/2b98d995192cc448af0499957635c5abbe8e4dea.

closed 2019-02-11 by jolivepetrus