whitecatboard/Lua-RTOS-ESP32 #225: pio.pin.setdir(pio.OUTPUT, 16) causes an immediate reboot

bytebang, 2019-02-09:
I try to compile LuaRTOS for the [TTGO-LORA32-v2](https://github.com/LilyGO/TTGO-LORA32-V2.0).
It compiles fine and I can confirm that the SD Reader and WIFI are working.

However - this module has also an 128x64 OLED display, which has to be be resetted (by pulling down PIN 16 to GND and setting it to HIGH again). Therefore one can use the following LUA commands:

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~{.lua}
display_rst = 16
pio.pin.setdir(pio.OUTPUT, display_rst)
pio.pin.setlow(display_rst)
pio.pin.sethigh(display_rst)
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

This procedure works on all other boards very well, but with this TTGOv2 board I am running into an issue where the GPIO16 can not be set as output pin - and therefore the OLED display can not be used.

The following snippet is made with `make monitor`, and it shows the issue:
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~{.lua}
Booting Lua RTOS...

  /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1549728121
commit 0f07c1e4932e9621f30231edbc409ac4b15fa9dc
Running from factory partition
board type GENERIC
cpu ESP32 rev 1 at 240 Mhz
spiffs start address at 0xf000, size 512 Kb
spiffs mounted on /

Lua RTOS beta 0.1 powered by Lua 5.3.4

/ > display_rst = 16
/ > pio.pin.setdir(pio.OUTPUT, display_rst)
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40152d20: 47e41c02 04b77744 84cab66b
0x40152d20: gpio_pullup_en at /home/user/esp-idf/components/driver/gpio.c:540 (discriminator 2)

Core 0 register dump:
PC      : 0x40152d24  PS      : 0x00060a30  A0      : 0x800ded16  A1      : 0x3ffd1da0  
0x40152d24: gpio_pullup_dis at /home/user/esp-idf/components/driver/gpio.c:540 (discriminator 2)

A2      : 0x3ffd1dc0  A3      : 0x00000010  A4      : 0x3ff4904c  A5      : 0x00010000  
A6      : 0x00000000  A7      : 0xff000000  A8      : 0x80153377  A9      : 0x3ffd1d70  
A10     : 0x00000010  A11     : 0x00000000  A12     : 0x00000010  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0x3ffba850  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x40152d24:0x3ffd1da0 0x400ded13:0x3ffd1dc0 0x400ecad3:0x3ffd1e00 0x400ecc1b:0x3ffd1e20 0x400ed099:0x3ffd1e50 0x400f5699:0x3ffd1e70 0x40102b29:0x3ffd1ea0 0x400f58f5:0x3ffd1ef0 0x400f5915:0x3ffd1f10 0x400fd7ad:0x3ffd1f30 0x400f52c1:0x3ffd1f50 0x400f5a9e:0x3ffd1fd0 0x400fe619:0x3ffd2000 0x400fd100:0x3ffd2030 0x400fd3de:0x3ffd2050 0x400fd625:0x3ffd2080 0x400f5699:0x3ffd20b0 0x400f58ee:0x3ffd20e0 0x400f5915:0x3ffd2100 0x400fd7ad:0x3ffd2120 0x400f52c1:0x3ffd2140 0x400f5a9e:0x3ffd21c0 0x400fe619:0x3ffd21f0 0x400fd394:0x3ffd2220 0x400d5adb:0x3ffd2250 0x400d565f:0x3ffd2270
0x40152d24: gpio_pullup_dis at /home/user/esp-idf/components/driver/gpio.c:540 (discriminator 2)

0x400ded13: gpio_pin_output_mask at /home/user/Lua-RTOS-ESP32/components/sys/drivers/gpio.c:1060

0x400ecad3: pio_op at /home/user/Lua-RTOS-ESP32/components/lua/modules/hw/pio.c:113

0x400ecc1b: pioh_set_pins at /home/user/Lua-RTOS-ESP32/components/lua/modules/hw/pio.c:211 (discriminator 1)

0x400ed099: pio_gen_setdir at /home/user/Lua-RTOS-ESP32/components/lua/modules/hw/pio.c:298
 (inlined by) pio_pin_setdir at /home/user/Lua-RTOS-ESP32/components/lua/modules/hw/pio.c:331

0x400f5699: luaD_precall at /home/user/Lua-RTOS-ESP32/components/lua/src/ldo.c:445

0x40102b29: luaV_execute at /home/user/Lua-RTOS-ESP32/components/lua/src/lvm.c:1275

0x400f58f5: luaD_call at /home/user/Lua-RTOS-ESP32/components/lua/src/ldo.c:510

0x400f5915: luaD_callnoyield at /home/user/Lua-RTOS-ESP32/components/lua/src/ldo.c:520

0x400fd7ad: f_call at /home/user/Lua-RTOS-ESP32/components/lua/src/lapi.c:1222

0x400f52c1: luaD_rawrunprotected at /home/user/Lua-RTOS-ESP32/components/lua/src/ldo.c:142 (discriminator 1)

0x400f5a9e: luaD_pcall at /home/user/Lua-RTOS-ESP32/components/lua/src/ldo.c:740

0x400fe619: lua_pcallk at /home/user/Lua-RTOS-ESP32/components/lua/src/lapi.c:1222

0x400fd100: docall at /home/user/Lua-RTOS-ESP32/components/lua/src/lua.c:258

0x400fd3de: doREPL at /home/user/Lua-RTOS-ESP32/components/lua/src/lua.c:423

0x400fd625: luaos_pmain at /home/user/Lua-RTOS-ESP32/components/lua/modules/sys/lua_adds.inc:228 (discriminator 1)

0x400f5699: luaD_precall at /home/user/Lua-RTOS-ESP32/components/lua/src/ldo.c:445

0x400f58ee: luaD_call at /home/user/Lua-RTOS-ESP32/components/lua/src/ldo.c:509

0x400f5915: luaD_callnoyield at /home/user/Lua-RTOS-ESP32/components/lua/src/ldo.c:520

0x400fd7ad: f_call at /home/user/Lua-RTOS-ESP32/components/lua/src/lapi.c:1222

0x400f52c1: luaD_rawrunprotected at /home/user/Lua-RTOS-ESP32/components/lua/src/ldo.c:142 (discriminator 1)

0x400f5a9e: luaD_pcall at /home/user/Lua-RTOS-ESP32/components/lua/src/ldo.c:740

0x400fe619: lua_pcallk at /home/user/Lua-RTOS-ESP32/components/lua/src/lapi.c:1222

0x400fd394: luaos_main at /home/user/Lua-RTOS-ESP32/components/lua/modules/sys/lua_adds.inc:258

0x400d5adb: lua_start at /home/user/Lua-RTOS-ESP32/components/sys/main.c:75 (discriminator 1)

0x400d565f: pthreadTask at /home/user/Lua-RTOS-ESP32/components/pthread/_pthread.c:551

Rebooting...
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

The mysterious thing is, that **the command `pio.pin.setdir(pio.OUTPUT, X)` works for all pins, but with pin 16 it crashes**. 

The display itself is OK, because as i received the board it had an other firmware flashed which displayed things correctly.

I have been trying now for more than 9 hours to fix this issue (e.g. by compiling an older version of LuaRTOS (Dec-2018), by manipulating the stack size, by enabling and disabling various settings, and so on) but now I am definitely running out of ideas. Has anybody experienced such a behavior , or can anybody point me into the right direction ?

jolivepetrus, 2019-02-11:
@bytebang,

Mmmmmm ......

This board use ESP32 PICO D4, and in this SOC GPIO16 is connected to the external SPI Flash. Usually it is not a good idea to use SPI Flash lines for other purposes.

See ESP32 PICO D4 datasheet:

https://www.espressif.com/sites/default/files/documentation/esp32-pico-d4_datasheet_en.pdf#page=18

Maybe using esp-idf and not Lua RTOS works, due to that portions of the program code are cached into RAM when you manipulate GPIO16, and maybe with Lua RTOS portions of code are not cached yet and because you manipulate SPI flash lines the code fails.

In any case if this board is using GPIO16 to reset the display it's a very, very bad idea.

bytebang, 2019-02-13:
Thanks for the reply.

It seems that a reset (with line 16) is not necessary. Unfortunately i have tried to get the display running on a board where the display is defect. On another unit the display works without the reset on the line 16.

Therefore i close this issue.

closed 2019-02-13 by bytebang