whitecatboard/Lua-RTOS-ESP32 #226: net.curl.get("http://example.com") doesn't work.

spierepf, 2019-02-13:
Installed using wcc on Ubuntu.

```
  /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1550053287
commit 70d4160f9ebc1de38ae716308204e28fc69d84f0
Running from ota_0 partition
board type DOIT-ESP32-DEVKIT-V1
cpu ESP32 rev 1 at 0 Mhz
flash EUI 304e36353117b246
spiffs start address at 0xf000, size 512 Kb
spiffs mounted on /

Lua RTOS beta 0.1 powered by Lua 5.3.4

Executing /system.lua ...
Starting wifi ...
Starting sntp client ...
Starting http server ...
http: server listening on port 80
Executing /autorun.lua ...

/ > net.curl("http://example.com")
stdin:1: attempt to call a nil value (field 'curl')
stack traceback:
        stdin:1: in main chunk
        [C]: in ?
/ > print(net.curl)
nil
/ > 

```

jolivepetrus, 2019-02-14:
@spierepf,

Same as https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/227, not included in official build, or default config for this board.

spierepf, 2019-02-14:
@jolivepetrus,

I am building my own firmware. I've wandered around the `make menuconfig` a bit, but could not figure out how to enable curl...

jolivepetrus, 2019-02-14:
@spierepf,

Yes, go to the following section in kconfig:

Component config > Lua RTOS > Lua > Lua RTOS Modules

Then select "Include curl support in net module", and finally build.

spierepf, 2019-02-14:
Recompiled with curl support, but now I get a new error:

```
Booting Lua RTOS...

  /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1550165014
commit 9c757a84beaa5e5cdd6a3605cb09de5d90bb59c1
Running from factory partition
board type DOIT-ESP32-DEVKIT-V1
cpu ESP32 rev 1 at 240 Mhz
flash EUI 304e36353117b246
spiffs start address at 0xf000, size 512 Kb
spiffs mounted on /

Lua RTOS beta 0.1 powered by Lua 5.3.4

Executing /system.lua ...
Executing /autorun.lua ...

/ > net.curl.get("http://example.com")
/home/developer/esp-idf/components/freertos/queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40097a94 on core 0

Backtrace: 0x40095003:0x3ffcf510 0x40095325:0x3ffcf530 0x40097a94:0x3ffcf550 0x401dbf32:0x3ffcf590 0x401dc1b8:0x3ffcf5b0 0x401ceff4:0x3ffcf5d0 0x401cf064:0x3ffcf5f0 0x401cb551:0x3ffcf610 0x401ca385:0x3ffcf630 0x401cdcaf:0x3ffcf670 0x4016dc65:0x3ffcf690 0x40153e47:0x3ffcf6b0 0x40153e7d:0x3ffcf6d0 0x400eeb33:0x3ffcf6f0 0x400f72c1:0x3ffcf760 0x40103509:0x3ffcf790 0x400f751d:0x3ffcf7e0 0x400f753d:0x3ffcf800 0x4010512d:0x3ffcf820 0x400f6ee9:0x3ffcf840 0x400f76c6:0x3ffcf8c0 0x40105f99:0x3ffcf8f0 0x40100c6c:0x3ffcf920 0x40100f4a:0x3ffcf940 0x40101191:0x3ffcf970 0x400f72c1:0x3ffcf9a0 0x400f7516:0x3ffcf9d0 0x400f753d:0x3ffcf9f0 0x4010512d:0x3ffcfa10 0x400f6ee9:0x3ffcfa30 0x400f76c6:0x3ffcfab0 0x40105f99:0x3ffcfae0 0x40100f00:0x3ffcfb10 0x400d5bd7:0x3ffcfb40 0x400d575b:0x3ffcfb60

Rebooting...
```

spierepf, 2019-02-14:
Sorry, I am an idiot. Didn't have wifi configured. Works fine.

closed 2019-02-14 by spierepf