whitecatboard/Lua-RTOS-ESP32 #227: Cannot mount a fat filesystem from SD card

spierepf, 2019-02-14:
I've followed the connection diagram at:

https://github.com/espressif/arduino-esp32/tree/master/libraries/SD

and can verify that running some of the SD arduino examples works as expected.

Perhaps I need to connect the SD card in a different way?

```
  /\       /\
 /  \_____/  \
/_____________\
W H I T E C A T

Lua RTOS beta 0.1. Copyright (C) 2015 - 2018 whitecatboard.org

build 1550101041
commit 9c757a84beaa5e5cdd6a3605cb09de5d90bb59c1
Running from ota_0 partition
board type DOIT-ESP32-DEVKIT-V1
cpu ESP32 rev 1 at 240 Mhz
flash EUI 304e36353117b246
spiffs start address at 0xf000, size 512 Kb
spiffs mounted on /

Lua RTOS beta 0.1 powered by Lua 5.3.4

Executing /system.lua ...
Executing /autorun.lua ...

/ > fs.mount("sd", "fat")
stdin:1: non-existent file system
stack traceback:
        [C]: in field 'mount'
        stdin:1: in main chunk
        [C]: in ?
/ > 

```

jolivepetrus, 2019-02-14:
@spierepf,

FAT it is not included into official builds for the board you have. In general, our criteria to include some feature into an official build is: if the board has a hardware component that needs a software piece, include it, so as this board does not have an SDCARD socket, FAT it is not included.

You can build Lua RTOS from the scratch and include FAT support.

It is not yet finished, but we are working on generating and automatic wiki page for each supported boards with which it is included in the official build. In the following link you have the info for your board:

https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/Firmware-DOIT-ESP32-DEVKIT-V1

spierepf, 2019-02-14:
@jolivepetrus,

Ok, cool. That makes sense.

Thanks!

closed 2019-02-14 by spierepf