whitecatboard/Lua-RTOS-ESP32 #228: ESP32-LyraT

ubaldus, 2019-02-22:
Hi, anyone already working on the ESP32-LyraT? 

If not, what would be the best way to add a new board?

I tried adding the relevant entries in:

components/sys/Kconfig
components/sys/luartos.h
boards/boards.json

and I created a boards/ESP32-LYRAT based on ESP32-WROVER-KIT but still when I select it from the menuconfig it does try to compile the "generic" version.

jolivepetrus, 2019-02-27:
@ubaldus,

We appreciate your interest in add support for new boards. Now we are made some refactory about adding support to new boards in an easy and agile way by the Lua RTOS community. I hope it will available in a few days.

jolivepetrus, 2019-02-28:
@ubaldus,

jolivepetrus, 2019-02-28:
@ubaldus,

@xopxe,

Please upgrade to https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/9efef9d273d0dd4f2fe5cf1df328b1f18e6f83c5, and follow the wiki: https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/How-to-add-a-new-board

closed 2019-02-28 by jolivepetrus