whitecatboard/Lua-RTOS-ESP32 #229: Support for Sparkfun Spark X board

xopxe, 2019-02-22:
We'd like to deply Lua-RTOS on these boards:

https://learn.sparkfun.com/tutorials/esp32-lora-1-ch-gateway-lorawan-and-the-things-network/all

Is support planned? We could try to port to this platform with some guidance.

jolivepetrus, 2019-02-27:
@xopxe,

We appreciate your interest in add support for new boards. Now we are made some refactory about adding support to new boards in an easy and agile way by the Lua RTOS community. I hope it will available in a few days.

jolivepetrus, 2019-02-28:
@xopxe,

Please upgrade to https://github.com/whitecatboard/Lua-RTOS-ESP32/commit/9efef9d273d0dd4f2fe5cf1df328b1f18e6f83c5, and follow the wiki: https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/How-to-add-a-new-board

closed 2019-02-28 by jolivepetrus