whitecatboard/Lua-RTOS-ESP32 #230: Wiki https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/NEOPIXEL-module

Mynogs, 2019-03-07:
Sample:
-- Setup a NEOPIXEL bus formed by 6 WS2812B leds, connected to GPIO14
neo = neopixel.**setup**(neopixel.WS2812B, pio.GPIO14, 6)  **<-- attache, not setup**

-- Set color for pixel 0 (pure red)
neo:setPixel(0, 255, 0, 0)

-- Set color for pixel 1 (pure green)
neo:setPixel(**0**, 255, 0, 0) **<-- 1 not 0**

-- Set color for pixel 2 (pure white)
neo:setPixel(**0**, 255, 255, 255) **<-- 2 not 0**

All sample use setup, must be attach

jolivepetrus, 2019-03-07:
@Mynogs,

I have just updated the wiki:

https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/NEOPIXEL-module

Can you review it?

Mynogs, 2019-03-07:
This is still wrong in all samples:
-- Set color for pixel 1 (pure green)
neo:setPixel(**1**, 255, 0, 0) <-- 1 not 0

-- Set color for pixel 2 (pure white)
neo:setPixel(**2**, 255, 255, 255) <-- 2 not 0

Mynogs, 2019-03-07:
Wrong button... sorry.

jolivepetrus, 2019-03-07:
@Mynogs,

Solved!

Mynogs, 2019-03-07:
Yes!

closed 2019-03-07 by Mynogs