whitecatboard/Lua-RTOS-ESP32 #231: Timer fires only once

Mynogs, 2019-03-07:
This is part of the p+ simulation system (Simulink like). 
This part execute the main simulation block. But the timer fires only one. What's wong with my code?

    if what == 'GENERATOR_MAIN' then
      return [[
  do
    print('Start p+ simulation ' .. (sim.stepRateS * 1000) .. 'ms cycle time')
    local t = tmr.attach(
      sim.stepRateS * 1000,
      function()  
        block.step()
        collectgarbage()
        sim.step = sim.step + 1
        sim.stepT0 = sim.stepT0 + 1
        sim.timeS = sim.timeS + sim.stepRateS
      end
    )
    t:start()
  end
      ]]
    end

Mynogs, 2019-03-07:
```
if what == 'GENERATOR_MAIN' then
  return [[
    do
      print('Start p+ simulation ' .. (sim.stepRateS * 1000) .. 'ms cycle time')
      local t = tmr.attach(
        sim.stepRateS * 1000,
        function()  
          block.step()
          collectgarbage()
          sim.step = sim.step + 1
          sim.stepT0 = sim.stepT0 + 1
          sim.timeS = sim.timeS + sim.stepRateS
        end
      )
      t:start()
    end
  ]]
```

jolivepetrus, 2019-03-07:
@Mynogs,

You missed to set which timer hardware to use. An example is as follows:

`
t = tmr.attach(
	tmr.TMR0,
	1000 * 50,
    function()  
	print("hi")
    end
)

t:start()
`

Review the attach's signature in the wiki:

https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/TIMER-module#instance--tmrattachid-period-callback

Mynogs, 2019-03-07:
I tryed to use the software timer.
Now I chanced it:
```
  do
    print('Start p+ simulation ' .. (sim.stepRateS * 1000) .. 'ms cycle time')
    local t = tmr.attach(
      tmr.TMR0,
      sim.stepRateS * 1000000,
      function()  
        block.step()
        collectgarbage()
        sim.step = sim.step + 1
        sim.stepT0 = sim.stepT0 + 1
        sim.timeS = sim.timeS + sim.stepRateS
      end
    )
    t:start()
  end
``` 
The timer still fired only once and the board no longer react to ctrl-c! I must press the reset button.
It is correct to end the Lua program? Or must I go to some kind of idle loop?

jolivepetrus, 2019-03-12:
@Mynogs,

In your tests, which is the value for sim.stepRateS?

jolivepetrus, 2019-03-16:
@Mynogs,

Not feedback about the value for sim.stepRateS. Please, reopen in case you need more assistance.

closed 2019-03-16 by jolivepetrus

Mynogs, 2019-03-16:
@jolivepetrus 
Sorry I missed your question!
The value of sim.simStepS is 0.1 
This means calculate a simulation step ever 100ms.