whitecatboard/Lua-RTOS-ESP32 #232: Set date and time

Mynogs, 2019-03-08:
I need to set (Lua) system time from a GPS receiver.
how to set the date and time of Lua?

the0ne, 2019-03-11:
@Mynogs currently it seems this is not implemented
Using which module do you contact the GPS receiver?

Mynogs, 2019-03-11:
@the0ne 
My first idea was to use the sensor "GPS". After I looked a little in the source code, I found that in nmea0183.c "nmea_GPRMC (char * sentence)" is disabled.
I would like to support the project, but have little time at the moment. So the GPS string will probably parse in Lua and will not change the system time.

the0ne, 2019-03-11:
Some interface could be added to set the system time from lua, maybe os.setgmttime(timestamp)
@jolivepetrus what do you think?

Mynogs, 2019-03-11:
That would be helpful!

jolivepetrus, 2019-03-11:
@Mynogs, @the0ne,

Sorry for the delay in my answer.

Well, there is a basic support for NMEA parser implemented in Lua RTOS, and GPS implemented in the sensor API. See:

https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/GPS-SENSOR
https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/components/nmea/nmea0183.c

In nmea0183.c the time is set in the nmea_GPRMC using the posix function mktime.

I think that it will be useful to implement a Lua function into os lib to set the current time in cases that user don't use GPS or don't use SNTP. Something like that:

os.settime(hours, minutes, seconds, month, day, year)

Mynogs, 2019-03-11:
@jolivepetrus, @the0ne 
Roberto uses tables in his date/time creation function.
Quote:
``
The first three fields are mandatory; the others default to noon (12:00:00) when not provided. In a Unix system (where the epoch is 00:00:00 UTC, January 1, 1970) running in Rio de Janeiro (which is three hours west of Greenwich), we have the following examples:

    -- obs: 10800 = 3*60*60 (3 hours)
    print(os.time{year=1970, month=1, day=1, hour=0})
      --> 10800
    print(os.time{year=1970, month=1, day=1, hour=0, sec=1})
      --> 10801
    print(os.time{year=1970, month=1, day=1})
      --> 54000   (obs: 54000 = 10800 + 12*60*60)
``
So replace os.time with os.settime. 
I would suggest doing that as well. I know this is a bit more work, but then it is consistent.

the0ne, 2019-03-11:
@Mynogs regarding your idea of time zones support:
As far as I know, there's no concept of using anything other than GMT in this OS.

Whenever _I_ need local time in a Lua-RTOS project, the lua code calculates whether it's currenly CET or CEST and then adds the corresponding offset to the system provided GMT time.

the0ne, 2019-03-11:
@Mynogs regarding
> I would suggest doing that as well. I know this is a bit more work, but then it is consistent.

What would it be consistent to? Is there a (Lua) standard for the os.time function other than it's available in Lua-RTOS? Can you provide a link there?

jolivepetrus, 2019-03-11:
@Mynogs, @the0ne,

In the past I renounced to support other that GMT due to if we want support for other timezones we will need a lot of space into flash. Almost this happened with older versions of Lua RTOS running on PIC32MZ.

About the os.time signature, for coherence with other Lua RTOS parts is better the proposed bellow:

os.settime(hours, minutes, seconds, month, day, year)

Only in GDisplay module tables are used to express coordinates, and this was done because usually in the school a point is represented by (x,y).

Mynogs, 2019-03-12:
@jolivepetrus @the0ne 
GMT is completely sufficient. The local time can usually be easily calculated.
My suggestion was to keep the os date / time functions consistent.
There are:
`t = os.time{year = 1970, month = 1, day = 1, hour = 0, sec = 1}`
So the set of time and date could look like this:
`os.settime(os.time{year = 1970, month = 1, day = 1 hour = 0 sec = 1})`
or
`os.settime{year = 1970, month = 1 , day = 1 hour = 0 sec = 1}`
Of course, if there are other spellings in other parts of Lua RTOS, it makes sense to keep them.

the0ne, 2019-03-12:
@jolivepetrus please check #246

jolivepetrus, 2019-03-12:
@Mynogs, @the0ne,

@the0ne, revised and merged. Thanks!.

the0ne, 2019-03-12:
@Mynogs you can use it both ways: either with parameters hours, minutes, seconds, month, day, year
or a table as with os.time

closed 2019-03-13 by jolivepetrus